GPSBabel/gpsbabel #1004: [Feature request] Exclude all track points outside a specified radius

turboscholz, 2023-02-13:
According to the [documentation](https://www.gpsbabel.org/htmldoc-development/filter_radius.html) one can filter out waypoints which are more away than a given radius limit to a given coordinate.

My use case is a bit different: I have a GPX file with several waypoints and a lot of trackpoints (with elevation information). The waypoints are located randomly distributed on top of the track. For some data analysis I need to filter out all trackpoints which are more away than a certain radius from a given waypoint coordinate.

As far as my tests have shown, the desired result is not possible with the current radius filter - after applying the filter, all waypoints outside of the limit are gone, but the complete track is still in the output GPX file. But I just need a small part of the track the coordinates of the one waypoint.

tsteven4, 2023-02-13:
Can you get what you want by using the https://www.gpsbabel.org/htmldoc-development/filter_nuketypes.html and https://www.gpsbabel.org/htmldoc-development/filter_transform.html filters along with the radius filter?

I.e., nuke the waypoints, transform the track to wpts and delete the track, run radius filter, transform wpts to trk and delete the wpts.

turboscholz, 2023-02-13:
Oh yes, that is a workaround, indeed! I was not aware of the nuketypes option. :+1:  The feature request can be closed from my point of view again. Thank you.

closed 2023-02-13 by turboscholz