GPSBabel/gpsbabel #1014: all_options.html: options all glued to following word

jidanni, 2023-03-01:
In [all_options.html](https://www.gpsbabel.org/htmldoc-development/all_options.html) the options,
```
    <p><code class="option">-w</code>Work o
    <p><code class="option">-T</code>Enable
    <p><code class="option">-b</code>Proces...
```
will all render glued to the following word. Please add one space in whatever
postprocessor you are using. Thanks.

And
```
-h -?Print help.
```
should not only be unglued:
```
-h -? Print help.
```
but also add a comma:
```
-h, -? Print help.
```

tsteven4, 2023-03-02:
I think this was all fixed in #941 10/30/2022.

@robertlipe , I would be happy to produce the gpsbabel.org target during CI and archive it.  This would alleviate the need to have all the necessary files for generation on ones personal machine.

tsteven4, 2023-03-07:
@jidanni Is there anything left here?  https://www.gpsbabel.org/htmldoc-development/all_options.html looks good to me.  I think the source was fixed previously as I mentioned and @robertlipe just need to get it to the server.

closed 2023-03-07 by jidanni