GPSBabel/gpsbabel #1015: fmt_unicsv.html: clarify "Add feet"

jidanni, 2023-03-01:
On https://www.gpsbabel.org/htmldoc-development/fmt_unicsv.html
please clarify
```
  alt =      Elevation (in meters) of the point. Add "ft" or "feet" for feet.
```
Say if you mean we should use
- alt ft, or
- altft, or
- ft

The first two give errors. So maybe you mean the last.
But then say "use instead" instead of "Add".

But saying just "ft" (or "feet") would then conflict with
```
prox =     Proximity (in meters).  Add "ft" or "feet" for feet.
```

So please add examples of what you mean to that page.

Also one notes the `-o unicsv` example has
```
No,Latitude,Longitude,Name,Date,Time,Filename
```
which means one cannot just go round
```
$ ... -o unicsv | ... -i unicsv
```
without shortening the header names back into their abbreviated form. Interesting.

tsteven4, 2023-03-02:
***alt ft***, ***altft*** both work, we try to be flexible.  In the example below an input altitude of 10 feet is converted to 3.048m in the gpx file (gpx ele element values are always in meters).

:~/work/vecs$ cat x.csv
No,Latitude,Longitude,Altitude,Date,Time,***alt ft***
1,51.599380,-177.425000,9754.0,2022/12/20,20:25:38,10

```
:~/work/vecs$ ./bld/gpsbabel -i unicsv -f x.csv  -o gpx -F -
<?xml version="1.0" encoding="UTF-8"?>
<gpx version="1.0" creator="GPSBabel - https://www.gpsbabel.org" xmlns="http://www.topografix.com/GPX/1/0">
  <time>2023-03-02T17:11:35.362Z</time>
  <bounds minlat="51.599380000" minlon="-177.425000000" maxlat="51.599380000" maxlon="-177.425000000"/>
  <wpt lat="51.599380000" lon="-177.425000000">
    <ele>3.048</ele>
    <time>2022-12-21T03:25:38Z</time>
    <name>WPT001</name>
    <cmt>WPT001</cmt>
    <desc>WPT001</desc>
  </wpt>
</gpx>
```

tsteven4, 2023-03-03:
This was resolved by #1018

tsteven4, 2023-03-06:
I mistakenly thought #1018 was closed.

tsteven4, 2023-03-07:
Closed, #1018 is merged.

closed 2023-03-07 by tsteven4