GPSBabel/gpsbabel #1016: [PATCH] Add example of using multiple filters

jidanni, 2023-03-01:
At the bottom of
https://www.gpsbabel.org/htmldoc-development/filter_interpolate.html
mention that sometimes one cannot use multiple filters in the same command.

```
Example of combining interpolate and time filters:
We must use two invocations:

gpsbabel -i kml -f i.kml -o kml,points=0,lines=0,track=1 \
-x interpolate,time=1 -F - |
gpsbabel -i kml -f -     -o kml,points=1,lines=0,track=0 \
-x track,start=$A,stop=$B -F o.kml
```

(P.S., I bet you no user could get the above just right like I did without days of trial and error, just like I did.)

)(No, I can't make a pull request, because the documentation system is so fragmented that there are several files involved all hard to track down which one is best, and also this is more than just a one line example, so you will have to be sure it looks good yourself because I don't have all that docbook or whatever stuff installed. So sorry it is not a real [PATCH].)

tsteven4, 2023-03-01:
Please provide a test case.  I can only guess this is an artifact of how kml maps to/from waypoints,routes and tracks.

jidanni, 2023-03-02:
[time2position.sh.gz](https://github.com/GPSBabel/gpsbabel/files/10870667/time2position.sh.gz)
[FlightAware_UAL871_KSFO_RCTP_20221220.kml.gz](https://github.com/GPSBabel/gpsbabel/files/10870668/FlightAware_UAL871_KSFO_RCTP_20221220.kml.gz)
Your goal is to get the same position as in [my investigation](https://commons.wikimedia.org/wiki/File:Eastern_part_of_Tanaga_Island_Alaska.jpg#Summary), but with only one invocation of gpsbabel, not two!

tsteven4, 2023-03-02:
I don't know what version you are using, and I'm guessing at your time, but the straightforward serial application of the interpolate and track filter seems to work fine.

$ PATH=./bld:$PATH ./t2p.sh FlightAware_UAL871_KSFO_RCTP_20221220.kml 20221221032718
{{Location|51.667003|-177.816329}}
$ PATH=./bld:$PATH ./time2position.sh FlightAware_UAL871_KSFO_RCTP_20221220.kml 20221221032718
{{Location|51.667003|-177.816329}}

t2p.sh:
```
# $0 FlightAware_UAL871_KSFO_RCTP_20221220.kml 20221221032538
gpsbabel -i kml -f $1 -x interpolate,time=1 -x track,start=$2,stop=$2 -o kml,points=1,lines=0,track=0 -F - | tac |
    perl -nwle'
        if (/coordinates>([^<]+)/) {
            @F = split ",", $1;
            printf "{{Location|%s|%s}}\n", @F[ 1, 0 ];
            exit;
        }
'
```

tsteven4, 2023-03-02:
You can also do this entirely in gpsbabel:
$ ./bld/gpsbabel -i kml -f FlightAware_UAL871_KSFO_RCTP_20221220.kml -x nuketypes,waypoints -x interpolate,time=1 -x track,start=20221221032538,stop=20221221032538 -o xcsv,style=x.style -F -
{{Location|51.599380|-177.425000}}

x.style:
```
# gpsbabel XCSV style file
DESCRIPTION     Comma separated values
#
# FILE LAYOUT DEFINITIIONS:
#
FIELD_DELIMITER         ""
RECORD_DELIMITER        NEWLINE
#
# INDIVIDUAL DATA FIELDS, IN ORDER OF APPEARANCE:
#
IFIELD CONSTANT, "{{Location|", "%s"
IFIELD LAT_DECIMAL,"","%f"
IFIELD CONSTANT, "|", "%s"
IFIELD LON_DECIMAL,"","%f"
IFIELD CONSTANT, "}}", "%s"
```

jidanni, 2023-03-03:
Well that is all amazing.
I'm not sure if the docs have any examples of combing two -x's.
Anyway maybe on the page I mentioned it might be useful
to add and example "Get location of airplane at time T from track"

gpsbabel -i kml -f flight_track.kml -x interpolate,time=1 -x track,start=$T,stop=$T -o kml,points=1,lines=0,track=0

(Note we never mentioned what if we wanted sub-second positions.)

jidanni, 2023-03-04:
I see, on the tips page there should be a new link, tips for air travelers. Or tips for aviation or tips for airplanes or whatever. Okay and then on that new page called air.html or whatever, there should be, "Junior mapper Dan Jacobson has submitted a tip about..

jidanni, 2023-03-04:
Okay, and then in my tip I'll make a story about how one was enjoying the hors d'oeuvres on a spanking new Boeing Dreamliner, when one realized that the electrostatic tinted glass windows blocked all GPS signals.

So how are we to 'geolocate' the photo that we took of the enemy spy base? The only thing useful in our Exif headers was just the time stamp. It was like we were using some camera from the turn of the century. You know 2000.

So where was that photo we took at 13:13:13 taken?...

jidanni, 2023-03-04:
Okay, now off the plane we had directly to the FlightAware etc websites where they offer one whole week of KML flight tracks at no charge.

Then we just set the start and stop times to be the same and then plug it into this nifty formula.. ~~and voila, we were at the South Pole. Haha. only joking. We were at the East pole.~~

Anyway so perhaps you guys at GPSbabel headquarters could make that HTML framework and I'll just come in and make some final adjustments to that air.html. Thanks!

GPSBabelDeveloper, 2023-03-04:
Remember to get that second photo with the label showing your sponsored
drink during hors d'oeuvre hour over the enemy base for your influencer
post. Please let your photograph be on your flight from Sydney to Santiago
that flies over Toronto that proves the electrostatic tinted glass windows
were provided by Big Globe™.

If you provide the prose for that, I'll absolutely make that part of the
GPSBabel canon. I've made everyone else mad at me today, so let's make it
everyone. Bring on the flat earthers from all over the globe!

(Limited time offer. Valid only to Dan.)

It's a bit funny that you mentioned EXIF support because just this week, I
was wondering if that's one of those things like the stack filter, sessions
support, or inifile support that seemed like a great nerd trick that got
put into our code but is never actually used by anybody in the real world.

jidanni, 2023-03-04:
All I know is I was using Exiftool to get the stuff out of the Exif headers.
Dreamliner nightmare: https://www.google.com/search?q=dreamliner+gps+signals+windows

jidanni, 2023-03-05:
OK, I made [my own tip page on my own website](https://www.jidanni.org/comm/air/routes/programs/gps/tips.html), so I'll close this issue. Thanks!

closed 2023-03-05 by jidanni