GPSBabel/gpsbabel #1019: First node of a track shouldn't have Heading. Or if it was a middle node before truncation, more could be mentioned

jidanni, 2023-03-05:
[We see](https://www.jidanni.org/comm/air/routes/programs/gps/tips.html) the Heading field is definitely bogus and should not be output on the first node of a trace.

But that first node could actually be fully filled out, as earlier in the same command line its heading and speed were indeed known.

Data file used:
[FlightAware_UAL871_KSFO_RCTP_20221220.kml.gz](https://github.com/GPSBabel/gpsbabel/files/10870668/FlightAware_UAL871_KSFO_RCTP_20221220.kml.gz)

tsteven4, 2023-03-05:
This is an ancient issue with track_recompute(), which is used by the kml, gpssim and gtrnctr **writers**.  I almost wonder if this was done on purpose for some unknown reason that is hard for us to imagine now.  A fix touches 9 reference files.  While it is fair game for a writer to manipulate output data as it sees fit, it is questionable practice for a writer to change the internal route and track data as track_recompute does.  This has side effects that will be visible to any subsequent writer.  An alternative is to use the track filter course and speed options to synthesize those fields.

To fill out the your selected first node add the course option when using the track filter as shown below.  This uses a different algorithm than track_recompute with it's own peculiarities related to the initial speed and course.  However since you aren't selecting the first point the anomalies aren't visible in your output.
```
$ t=20221221032538 f=FlightAware_UAL871_KSFO_RCTP_20221220.kml
$ [gpsbabel](https://www.gpsbabel.org/) -i kml -f $f \
-x interpolate,time=1 \
-x track,course,start=$(($t-1)),stop=$t_ \
-o kml,units=a,points=1,lines=0,track=0 -F out.kml
```

jidanni, 2023-03-06:
@tsteven4 Sorry to ruin your day but, no difference observed:

t=20221221032538 f=FlightAware_UAL871_KSFO_RCTP_20221220.kml;\
for i in "" ,course; do echo ===$i; \
gpsbabel -i kml -f $f \
 -x interpolate,time=1 -x track$i,start=$(($t-1)),stop=$(($t)) \
 -o kml,units=a,points=1,lines=0,track=0 -F -|grep -vi snippet.created|sum; done
===
35157     6
===,course
35157     6

tsteven4, 2023-03-06:
Kill gpssim.
Keep gtrnctr, that is an active format, I even use it personally.

tsteven4, 2023-03-06:
Seems good to me:
```
t=20221221032538 f=FlightAware_UAL871_KSFO_RCTP_20221220.kml;for i in "" ,course; do echo ===$i; ./gpsbabel -i kml -f $f  -x interpolate,time=1 -x track$i,start=$(($t-1)),stop=$(($t))  -o kml,units=a,points=1,lines=0,track=0 -F out.$i; done; diff out. out.,course
===
===,course
130a131
> <tr><td>Heading: 285.7</td></tr>
```

tsteven4, 2023-03-07:
> Seems good to me:

Ahh. I had a couple changes involved.
1) don't recompute Heading if the wpt already has one.
2) don't compute Heading for the first point.

the course option takes path 1) and computes 285.7 for the first point.  Without it you take path 2) and avoid computing the erroneous 358.0.  But you don't have either change.

jidanni, 2023-03-07:
No big deal, I was just making some holiday computations. I'll stick with Debian sid.

GPSBabelDeveloper, 2023-03-07:
Debian updates from us extremely rarely.

Then they remove the update check just to be extra sure you're out of date.
:-(

jidanni, 2023-03-07:
It seems each upstream should send a message to each maintainer at each distribution, telling them when a new version is ready.
Seems much smarter then each maintainer every once in a while looking around to see if there are any new versions.

tsteven4, 2023-03-07:
I believe Debian monitors our repo to discover changes.  That is the purpose of https://salsa.debian.org/debian-gps-team/pkg-gpsbabel/-/blob/master/debian/watch.  Many distros do something similar.

1.8.0 has made it into sid(unstable) and bookworm(testing), but not  bullseye(stable).  I'm not sure what the hangup is.  The status is here https://tracker.debian.org/pkg/gpsbabel

tsteven4, 2023-03-07:
Fixed in #1031

closed 2023-03-07 by tsteven4