GPSBabel/gpsbabel #1023: -x interpolate,time= Sub second interval wavering

jidanni, 2023-03-06:
Here we have requested an interval on 0.3 seconds.
Normally this should return to a whole number of seconds every
10 * .3 = 3 seconds. But sometimes it "wavers":
```

$ gpsbabel -i kml -f $f -x interpolate,time=0.3 \
-o kml,units=a,points=1,lines=0,track=0 -F -|
perl -lnwe 'BEGIN{$m=0} next unless /when/;
if (/:..Z/){printf "\n$_: $m"; $m=0;}
else { $m++;}'|tail -n 33|sed 's/ *//'
<when>2022-12-21T11:27:46Z</when>: 9
<when>2022-12-21T11:27:49Z</when>: 9
<when>2022-12-21T11:27:52Z</when>: 9
<when>2022-12-21T11:27:55Z</when>: 9
<when>2022-12-21T11:27:58Z</when>: 9
<when>2022-12-21T11:28:01Z</when>: 9
<when>2022-12-21T11:28:04Z</when>: 9
<when>2022-12-21T11:28:07Z</when>: 9
<when>2022-12-21T11:28:10Z</when>: 9
<when>2022-12-21T11:28:13Z</when>: 9
<when>2022-12-21T11:28:32Z</when>: 63
<when>2022-12-21T11:28:40Z</when>: 26
<when>2022-12-21T11:28:48Z</when>: 26
<when>2022-12-21T11:28:56Z</when>: 26
<when>2022-12-21T11:29:04Z</when>: 26
<when>2022-12-21T11:29:35Z</when>: 103
<when>2022-12-21T11:29:38Z</when>: 9
<when>2022-12-21T11:29:41Z</when>: 9
<when>2022-12-21T11:29:44Z</when>: 9
<when>2022-12-21T11:29:47Z</when>: 9
<when>2022-12-21T11:29:50Z</when>: 9
<when>2022-12-21T11:29:53Z</when>: 9
<when>2022-12-21T11:29:56Z</when>: 9
<when>2022-12-21T11:29:59Z</when>: 9
<when>2022-12-21T11:30:02Z</when>: 9
<when>2022-12-21T11:30:05Z</when>: 9
<when>2022-12-21T11:30:25Z</when>: 66
<when>2022-12-21T11:30:57Z</when>: 106
<when>2022-12-21T11:31:29Z</when>: 106
<when>2022-12-21T11:31:37Z</when>: 26
<when>2022-12-21T11:31:45Z</when>: 26
<when>2022-12-21T11:31:53Z</when>: 26
<when>2022-12-21T11:32:01Z</when>: 26
```
Yes, this is no big deal, but weird.

tsteven4, 2023-03-06:
I think you are noticing that the output of the interpolate filter is not sampled at a constant rate.

The promise of the interpolate filter is "This filter modifies any tracks so that either the distance or the time between consecutive points **is no more than** the specified interval.".  Specifically, it does not promise to sample the track at a constant sample rate or distance.

The implementation actually inserts just enough points between every pair of existing points that the promise is kept.  If the input data is not regularly sampled, then the output may not be regularly sampled.  This is an implementation detail, i.e. it isn't promised that this is how the filter will work.

tsteven4, 2023-03-14:
Closing.  The interpolate filter time option does not promise a constant sample rate, just a maximum sample period between points.

closed 2023-03-14 by tsteven4