GPSBabel/gpsbabel #1024: End artificial start=, stop= whole second restriction

jidanni, 2023-03-06:
Here our track is sliced into nice half second segments,
but due to the "artificial" restriction of start= and stop= being whole
seconds, we encounter:
```
-x interpolate,time=0.5 -x track,start=20221221032537.5,stop=20221221032538.5
trackfilter-range-check: Invalid value for option "20221221032537.5"!
```

I mean imagine it's year 2222 and everybody is zipping around real fast.
Traffic tickets are being issued for microsecond violations and
gpsbabel is still thinking in seconds.

In real life I can [query the position of a plane at a particular second](https://www.jidanni.org/comm/air/routes/programs/gps/tips.html).
But my cellphone camera's Exif also have the microsecond,
and airplanes travel many meters per second,
so the only block on me being able to get a more precise answer
for where the plane was is this artificial restriction.

Anyway please allow
`-x interpolate,time=A.B -x track,start=C.D,stop=E.F`
where all six need not be related.
I.e., the two terminal points need not be on a node,
but should still have the correct position at that point.
Yes the user might get irregularly  spaced nodes
`o-o---o---o---o---o---o---oo `at the ends, but that is what he requested
via the start=, stop= parameters.

tsteven4, 2023-03-06:
The interpolate filter has supported subsecond interpolation since 1.7.0.  As I have explained in #1023 it does not guarantee an exact sample rate.

The track filter start and stop times are currently restricted to integral seconds.

At best we support millisecond resolutions, not microseconds.  This is the resolution supported by [QTime](https://doc.qt.io/qt-6/qtime.html), which we use to store all our time data.

The track filter does not generate points at the stop and start times.  the start/stop operation just selects a set of existing points.  if a point exists at the start(stop) time then it is included.

tsteven4, 2023-03-06:
#1026 adds millisecond resolution for the track filter start and stop options.

tsteven4, 2023-03-07:
@jidanni The change has been merged.  Can you build and try it?

jidanni, 2023-03-08:
Erg... no worries, in a couple years it will get to Debian, and if I'm not in the Old Folks home and had my porridge...

Anyways, I'll trust you! Besides, what if I found another bug trying, that would only increase your work :->

Anyways, don't worry, I'm sure it works. You see I already labeled all my hobby pics, so it's on to the next adventure.

closed 2023-03-08 by jidanni