GPSBabel/gpsbabel #1040: Allow directly setting KML altitudeModes

jidanni, 2023-03-24:
Allow directly setting KML altitudeModes

On https://www.gpsbabel.org/htmldoc-development/fmt_kml.html
>floating option
>Altitudes are absolute and not clamped to ground.
>When this option is nonzero, altitudes are allowed to float above or below the ground surface. By default, this option is zero so that altitudes are clamped to the ground. Specify floating=1 to allow them to float.

OK, that's great, but why not also give full control to the user to
pick the many possible allowed KML `altitudeMode`s?

That way he needn't post process the KML file with e.g.,
```
for($dom->findnodes('//altitudeMode')){
    $_->unbindNode for $_->childNodes();
    $_->appendText('relativeToGround');
}
```

GPSBabelDeveloper, 2023-03-24:
This was all that existed when this flag was added.  "Floating" was so that
recorded flight tracks showed a plane's elevation. (There was an open
request for "floating, except don't go under the ground" so that landings
and takeoffs didn't disappear underground, but I think that never took off.
(Ha!) Still, There are only three such states in IGC KML today:

clampToGround, relativeToGround, absolute

Most GPS-like systems where you want to draw data on a map and want the
data either on the ground, regardless of what the (notoriously inaccurate)
GPS alt says, above the ground, or above MSL I don't think we cover that
middle ground. That's not "many".

There are some gx:extensions that were introduced during the Ocean effort
in Earth 5, I think, allowing distances relative to sea floor (for an item
neither floating nor totally sunken) and one that was initially internally
called "clamped to crust" for objects at rest on water's bottom.

It's unfortunate that in a KML->KML conversion we damage this, but we
mangle a LOT of KML in this way and haven't incurred much grief over it.

Maybe instead of 'floating' being a bool we could make it effectively an
enum {0, 1, 2} with 2 being relativeToGround. That wouldn't balloon our
UI/doc. Does that get you to a happy place?

RJL

jidanni, 2023-03-25:
Well I didn't check what these all do
[image]
and maybe more will be on the way. So I was just thinking of a maintenance free solution... but some of them would need a gx: in the front... so one cannot win...