GPSBabel/gpsbabel #1047: 1x1 meter square not big enough for -o osm ?

jidanni, 2023-03-26:
Here I give `-o osm` a 1x1 meter object.
Well, yes, it makes a closed way. Good.
However all the nodes are the same node!
Is 1x1 meter "not big enough for its little brain?" :-)

```
gpsbabel -i kml -f - -o osm,created_by=  -F -
<?xml version='1.0' encoding='UTF-8'?>
<osm version='0.6' generator='GPSBabel-1.8.0'>
  <node id='-1' visible='true' lat='24.1817855' lon='120.8641815'>
  </node>
  <way id='-2' visible='true'>
    <nd ref='-1'/>
    <nd ref='-1'/>
    <nd ref='-1'/>
    <nd ref='-1'/>
    <nd ref='-1'/>
    <tag k='name' v='tank'/>
  </way>
</osm>
```
Here's the Placemark in question:
```
<Placemark>
  <name>tank</name>
  <description>1m x 1m = 1 m2</description>
  <Polygon>
    <outerBoundaryIs>
      <LinearRing>
        <coordinates>120.8641815,24.1817855 120.8641815,24.1817765 120.8641725,24.1817765 120.8641725,24.1817855 120.8641815,24.1817855</coordinates>
      </LinearRing>
    </outerBoundaryIs>
  </Polygon>
</Placemark>
```

jidanni, 2023-03-26:
Maybe have an environment variable where people could increase the sensitivity without it wrecking it for others.

You got a whole field of catastral mapping that you're locking the application out of.

jidanni, 2023-03-26:
Actually, if you think about it, here we're trying to perform a very valuable task. It doesn't involve interpolating the numbers at all. It's just rearranging them into a different format.

It's almost as if we need to encode our numbers or rot13 (probably wouldn't work) to keep the program from thinking too much about them.

Therefore there needs to be a mode, that just simply disengages any introspection of the numbers at all, allowing rearrangement of them, unhindered.

jidanni, 2023-03-27:
https://github.com/GPSBabel/gpsbabel/issues/1048#issuecomment-1484001531
> Six digits gets below .4mm, which is on on the order of the thickness of a nice business card. I can live with that.

Me too, but -o osm was blowing up at even one meter,
```
$ perl -wle 'printf "%.9f\n", 24.1817855 - 24.1817765;'
0.000009000
$ perl -wle 'printf "%.9f\n", 360/40000000;'
0.000009000
```
So something indeed is wrong.

tsteven4, 2023-03-27:
OsmFormat::osm_name_from_wpt uses QString::arg(double) with one argument, which by default uses format specifier 'g' with a precision of -1.  QLocale shows format specifier 'g' interprets precision as the "maximum number of significant digits (trailing zeroes are omitted)", and " any negative precision is ignored in favor of the default, 6."  Thus OsmFormat::osm_name_from_wpt may not get us to six digits after the decimal point.

This can be verified by with the example from #1047 using qDebug to print the names:
"\u000124.1818\u0001120.864"
"\u000124.1818\u0001120.864"
"\u000124.1818\u0001120.864"
"\u000124.1818\u0001120.864"
"\u000124.1818\u0001120.864"

1-e6 degrees is 0.11 meters at the equator, not 0.4mm as stated in #1048.

I would suggest explicitly setting the fieldwidth, format and precision to 0, 'f', and 6 respectively when converting latitude and longitude in osm_name_from_wpt.

tsteven4, 2023-03-27:
As a workaround you can use the transform filter, which will have the side effect of naming the points.  This will cause the osm writer to recognize them as being different.  However, it will also cause the first and last point in the LinearRing to be transformed into different nodes.
./bld/gpsbabel -t -i kml -f x.kml -x transform,rte=trk,del -o osm -F -

PR #1053 (2023-03-31): use 6 digits after the decimal point for osm_name_from_wpt.
I beleive this restores the behavior from before commit 4673a25, August 4, 2013.

This should resolve #1047.
commit 978352c0 (2023-03-31): use 6 digits after the decimal point for osm_name_from_wpt. (#1053)

I beleive this restores the behavior from before commit 4673a25,
August 4, 2013.

This should resolve #1047.
closed 2023-03-31 by tsteven4