GPSBabel/gpsbabel #1050: Idea: PROLOGUE AUTO

jidanni, 2023-03-26:
Regarding [PROLOGUE](https://www.gpsbabel.org/htmldoc-development/style_layout.html), I got an idea!

- PROLOGUE AUTO

Which would mean to generate the prologue automatically, based on the
IFIELDs and FIELD_DELIMITER.

For instance,

```
RECORD_DELIMITER NEWLINE
PROLOGUE AUTO
FIELD_DELIMITER COMMA
IFIELD LAT_DECIMAL,,%f
IFIELD LON_DECIMAL,,%f
IFIELD GMT_TIME,,%H:%M:%S
IFIELD ALT_FEET,,%.0f
IFIELD PATH_SPEED_KPH,,%.1f
IFIELD PATH_COURSE,,%f
```
would be equivalent to

```
RECORD_DELIMITER NEWLINE
PROLOGUE LAT_DECIMAL,LON_DECIMAL,GMT_TIME,ALT_FEET,PATH_SPEED_KPH,PATH_COURSE
FIELD_DELIMITER COMMA
IFIELD LAT_DECIMAL,,%f
IFIELD LON_DECIMAL,,%f
IFIELD GMT_TIME,,%H:%M:%S
IFIELD ALT_FEET,,%.0f
IFIELD PATH_SPEED_KPH,,%.1f
IFIELD PATH_COURSE,,%f
```

but without the user needing to type everything twice, or often forgetting to
update the PROLOGUE every time he adds an IFIELD.

Sure, he could still do it the old way, if he doesn't want SCARY CAPS.
But when the result is e.g., going to be consumed by sqlite etc., why
force him to come up for names of fields that already have names?

You will note that I placed PROLOGUE AUTO before FIELD_DELIMITER,
that way the program can start gathering the information it needs at
the right time.

Yes, call it AUTOMATIC instead of AUTO. Whatever. Thanks.

jidanni, 2023-03-26:
No I think you are missing my point entirely.

I am saying style files are great.

They could be greater if they had one feature.

The feature is a way to keep the PROLOGUE line consistent with the fields one has inserted below it.

Just a way to make it simpler to maintain style files.

For those users who want PROLOGUEs that simply reflect the fields further down in the file.

Also, I want to add to my idea:

Here's how one could get arbitrary titles too:

Taking a idea from SQL: "AS":

` IFIELD PATH_COURSE,,%f AS Bearing`

This would put `Bearing` instead of `PATH_COURSE` into the header!

GPSBabelDeveloper, 2023-03-27:
unicsv effectively writes the prologue for you.

Most people are writing files that need specific ordering. I don't think we
can dlegate that to auto.

I do geneerally think that styles are one of the best ideas we'e had, but
it's not like we have (or have ever had) people writing tons of style files.

jidanni, 2023-03-27:
@GPSBabelDeveloper thanks, but for starters: #1042 .