GPSBabel/gpsbabel #1098: Problem converting kml with large number of coordinates to osm

krzyk, 2023-05-02:
I have a kml file that has a bit of coordinates ([sq.zip](https://github.com/GPSBabel/gpsbabel/files/11374511/sq.zip)) and I would like to convert it to osm.
I tried following command line:
```
gpsbabel -i kml -f sq.kml -o osm,tag=highway:primary -F tiles.osm
```

But all I get is an almost empty tiles.osm file with just the main tag.
I tried with `-D 5` option (the docs miss what values can be put there so I tried with `5`), but didn't get any errors, I'm not sure what I'm doing wrong, or such file is maybe unsupported?

When I upload that kml file to online tools it shows the correct data on maps.

krzyk, 2023-05-02:
@tsteven4 Thanks, this looks much better, but the shape doesn't have holes, which are there in the kml file.

I checked kml file in google earth and osm file in JOSM.

tsteven4, 2023-05-02:
yes, we ignore the innerBoundaryIs/LinearRing elements.

We support waypoints, routes and tracks, not arbitrary GIS shapes.  The outerBoundaryIs/LinearRing elements are mapped to tracks.

krzyk, 2023-05-02:
That one looks great. Thanks.

I use JOSM to view osm files.

PR #1099 (2023-05-03): use QRegularExpression for xml generic tag matching.
The kml reader now creates tracks from innerBoundryIs elements in addition to the historic creation from outerBoundryIs elements.

This resolves #1098.
closed 2023-05-03 by tsteven4