GPSBabel/gpsbabel #1101: KML Placemark-elements with MultiGeometry are ignored

velle, 2023-05-07:
Im trying to translate a kml file to osm. I kept getting osm files with no features, and it took me a while to realize why. 

```
gpsbabel -i kml -f input.kml -o osm -F output.osm
````

### Input KML

```
<?xml version="1.0" encoding="utf-8" ?>
<kml xmlns="http://www.opengis.net/kml/2.2">
<Document id="root_doc">
    <Placemark>
      <name>alpha</name>
      <MultiGeometry><Point><coordinates>10.0000,56.0000</coordinates></Point></MultiGeometry>
    </Placemark>
    <Placemark>
      <name>beta</name>
      <Point><coordinates>11.0000,57.0000</coordinates></Point>
    </Placemark>
</Document>
</kml>
```

### Output OSM

```
<?xml version='1.0' encoding='UTF-8'?>
<osm version='0.6' generator='GPSBabel-1.8.0'>
  <node id='-1' visible='true' lat='57.0000000' lon='11.0000000'>
    <tag k='created_by' v='GPSBabel-1.8.0'/>
    <tag k='name' v='beta'/>
    <tag k='note' v='beta'/>
  </node>
</osm>
```

### Qgis produces KML with Multigeometry ###

I have only tried exporting from WFS to KML, and I have only tried with Points and MultiPoints. If exporting a WFS-layer with geometry type Point, the output KML does not use MultiGeometry. But if exporting from a WFS-layer having geometry type MultiPoint, even if all exported features have exactly one point and even if explicitly choosing the output geometry type to be Point, it does use MultiGeometry in the output OSM. 

Im not sure if this is intended behavior in QGIS. But in any case it would be great if gpsbabel could still handle such data as input. Or at least give a warning or error. 

### Questions

1) Is it intended behavior that gpsbabel ignores these?

2) Is it intended behavior that gpslabel does not even produces a warning or notification that some Placemarks are being ignored?

3) Would it be possible to change this behavior so that gpsbabel produces a warning and/or actually does translate placemarks when they use MultiGeometry (obviously only in the cases where possible; I don't know much about the limitations of the OSM format).

tsteven4, 2023-05-07:
There was a related issue a few days ago.  It was fixed in #1099.  Either build from source, or for the adventurous you can try one of our latest [continuous integration pre-release builds](https://github.com/GPSBabel/gpsbabel/releases/).  Does this take care of all your concerns?

```
$ ./bld/gpsbabel -i kml -f x.kml -o osm -F -
<?xml version="1.0" encoding="UTF-8"?>
<osm version="0.6" generator="GPSBabel-1.8.0">
  <node id="-1" visible="true" lat="56.0000000" lon="10.0000000">
    <tag k="created_by" v="GPSBabel-1.8.0"/>
    <tag k="name" v="alpha"/>
    <tag k="note" v="alpha"/>
  </node>
  <node id="-2" visible="true" lat="57.0000000" lon="11.0000000">
    <tag k="created_by" v="GPSBabel-1.8.0"/>
    <tag k="name" v="beta"/>
    <tag k="note" v="beta"/>
  </node>
</osm>
```

velle, 2023-05-07:
Wow that was fast :) Thanks. 

I forgot to include version number in my original post, but I just edited to include that. 

I am currently trying to see if I can compile from source, but with my lack of experience I haven't succeeded yet. I will let you know what I find out :)

tsteven4, 2023-05-07:
build dependencies for ubuntu in tools/Dockerfile_jammy (or use download the image from https://hub.docker.com/r/tsteven4/gpsbabel_build_environment_jammy/tags), then tools/build_and_test_cmake.sh.

velle, 2023-05-07:
Yes, in fact that did solve my problem. With the version I just compiled, the output osm DOES include both placemarks :) Thanks. 

```
<?xml version="1.0" encoding="UTF-8"?>
<osm version="0.6" generator="GPSBabel-1.8.0">
  <node id="-1" visible="true" lat="56.0000000" lon="10.0000000">
    <tag k="created_by" v="GPSBabel-1.8.0"/>
    <tag k="name" v="alpha"/>
    <tag k="note" v="alpha"/>
  </node>
  <node id="-2" visible="true" lat="57.0000000" lon="11.0000000">
    <tag k="created_by" v="GPSBabel-1.8.0"/>
    <tag k="name" v="beta"/>
    <tag k="note" v="beta"/>
  </node>
</osm>
```

tsteven4, 2023-05-07:
This was fixed in #1099

closed 2023-05-07 by tsteven4