GPSBabel/gpsbabel #1105: sub-second timestamps in OSM XMLs

alexanderzatko, 2023-05-12:
I used gpsbabel to convert a gpx file containing time elements with sub-second timestamps, like this:

`
... 
<trkpt lat="48.518543" lon="19.0508291">
        <ele>457.251</ele>
        <time>2023-04-29T11:06:22.029Z</time>
</trkpt>
...
`

Gpsbabel dutifully copies the timestamps to an OSM XML when asked:

`<osm version='0.6' generator='GPSBabel-1.6.0'>
  <node id='-1' visible='true' lat='48.5185430' lon='19.0508291' timestamp='2023-04-29T11:06:22.029Z'>
    <tag k='created_by' v='GPSBabel-1.6.0'/>
  </node>
...
`

However, I am [told](https://github.com/osmcode/osmium-tool/issues/267) that OSM files never contain subsecond timestamps and feeding such files to applications like osmosis leads to errors.

tsteven4, 2023-05-12:
gpsbabel 1.6.0 is no longer maintained, but the latest release 1.8.0 also writes subsecond timestamps in osm files.

subsecond timestamps in osm files appear to be allowed.

https://wiki.openstreetmap.org/wiki/Elements shows an example with subsecond times and cites a reference that supports them:

<html>
<body>

timestamp | W3C standard date and time formats. | Time of the last modification (e.g. "2016-12-31T23:59:59.999Z").
-- | -- | --

</body>
</html>

tsteven4, 2023-05-12:
Also  this https://wiki.openstreetmap.org/wiki/API_v0.6/DTD has a comment refering to ISO 8601, which allows subseconds.
```
<!ATTLIST node timestamp         CDATA #IMPLIED><!-- ISO 8601 format -->
```

joto, 2023-05-12:
Unfortunately the OSM data formats are not very well and very strictly defined. While technically this might be allowed by those definitions, in practice none of the usual software generating OSM files generates sub-second timestamps, because internally the OSM database doesn't support them. The other widely used OSM file format, the [PBF Format](https://wiki.openstreetmap.org/wiki/PBF_Format) definitely doesn't allow sub-seconds, because it stores the seconds in an integer.

So creating OSM files with sub-second granularity is problematic. Some software does not support them at all, some might, some might seems to support it but truncate the data internally in some way. So I recommend not creating files with sub-seconds to allow for maximal compatibility.

Btw: There is a similar issue with time zones. OSM times are always in UTC and `Z` ist the only time zone designation supported everywhere in OSM files. ISO 8601 allows for different time zones, but OSM file don't.

closed 2023-05-12 by tsteven4

tsteven4, 2023-05-12:
@alexanderzatko you may try one of our [pre-release "Continuous builds"](https://github.com/GPSBabel/gpsbabel/releases) for windows or macos, or build from source.  If you can please verify your issue is solved.

alexanderzatko, 2023-05-12:
Thank you. I will try to build, as I am on ubuntu. I will report back when done.

tsteven4, 2023-05-12:
some places to look for dependencies and our build script:
https://github.com/GPSBabel/gpsbabel/issues/1101#issuecomment-1537498678

mmd-osm, 2023-05-13:
> because internally the OSM database doesn't support them

The OSM database (apidb) itself uses the standard PostgreSQL datatype _timestamp_ which has microseconds resolution. However, every consumer of that information discards any sub-second information, and as @joto has pointed out, pretty much all file formats to exchange osm data don’t support it.

tsteven4, 2023-05-13:
@alexanderzatko There is now a [pre-release "Continuous build" for Linux](https://github.com/GPSBabel/gpsbabel/releases/tag/Continuous-Linux).  It needs to be installed with devmode and dangerous, e.g. "sudo snap install --dangerous --devmode gpsbabel_LinuxInstaller-9dccbfc-20230513_amd64.snap".  It only contains the command line client gpsbabel, there the GUI is not included as of yet.

alexanderzatko, 2023-05-13:
Great, the installation process was flawless and my gpx file was converted with second timestamps. Thank you.

PR #1107 (2023-05-12): restrict osm timestamps.
osm consumers may not accept fractional seconds, but all should support whole seconds.
osm consumers may not accept time zone designations other than Z, but all should support time zone designation Z.
See https://github.com/GPSBabel/gpsbabel/issues/1105

This resolves #1105