GPSBabel/gpsbabel #1115: Converting encoding of waypoints names

svintuss, 2023-05-15:
Hello. 

I'm trying to use GPSBabel for importing and exporting tracks and waypoints from/to Garmin GPSmap 60cx. As other older units it doesn't support unicode. 
Is it possible to make on-the-fly conversion of names and comments from 8-bit encoding to UTF-8 and vice-versa? E. g. from Win-1251 to UTF8?

So far I'm getting waypoints like these in .gpx:
```
  <wpt lat="57.142150495" lon="33.117158515">
    <name>5 ÊÀËÈÍÈÍÀ</name>
    <cmt>5 ÊÀËÈÍÈÍÀ</cmt>
    <desc>5 ÊÀËÈÍÈÍÀ</desc>
    <sym>Lodging</sym>
  </wpt>
```

tsteven4, 2023-05-15:
What version of gpsbabel?

What platform?  windows, linux, macos?

What is the command line that you are using?

We do have this comment in garmin.cc that implies this might have been broken years ago.  It is difficult to maintain this module without a battery of devices to test against, and such testing is manual.  Robert will tremble, but it does look possible to try to use the correct codec, but we would require help with testing.
```
  /*
   * This used to mean something when we used cet, but these days this
   * format either uses implicit QString conversions (utf8),
   * or we have hard coded QString::fromLatin1, CSTRc, or CSTR.  These
   * are likely bugs.
   * However, this is still used for garmin_fs_garmin_after_read,
   * garmin_fs_garmin_before_write.
   */
```

tsteven4, 2023-05-15:
Can you build the code yourself?

svintuss, 2023-05-15:
Yes, I can.

tsteven4, 2023-05-15:
Hurray.  glad to have help.  I'll push something up that is a start in a bit.

tsteven4, 2023-05-15:
You can try this [branch](https://github.com/GPSBabel/gpsbabel/tree/garmincodec).

I think it will fix the garmin reader.  It doesn't touch the garmin writer, so that will still be broken.  Add option -vs to get verbose status which should print a message "reciever charset detected as ..." to stdout when you try to use the garmin format to read from your device.  To start read from the device and write to gpx, let's see if that looks better.

svintuss, 2023-05-15:
Thanks again. After building the `garmincodec` branch I have the charset detected as windows-1252. That should be technically correct, but I thought of manually specifying encoding. E. g. "source encoding" and "destination encoding" options like`-se windows-1251` and `-de UTF8`.

```
% ./gpsbabel -vs -w -i garmin -f usb: -o gpx -F test1.gpx
reciever charset detected as windows-1252.
100/46/46
100/46/46
```

tsteven4, 2023-05-15:
There currently isn't a way to override the encoding.  Based on the type of device we assign a character set.  Does the device have support for more than one character set?

how does the gpx look?  it should be correctly encoded in utf-8.

svintuss, 2023-05-15:
No, these Garmins are from pre-unicode era, they only support windows-1252. The GPX produced is correctly encoded in utf8 but extended latin characters are treated as extended latin and not cyrillic. 

I thought of a workaround when I open the .gpx file > save it with Latin1 encoding > open with windows-1251 encoding > save again with utf8.

tsteven4, 2023-05-15:
Hmmm.  so the device only supports 1252 but your using it with 1251?

svintuss, 2023-05-15:
Yes, that's how old devices were localised. Extended latin font characters were replaced with Cyrillic counterparts and MapSource was forced to use cp1251 instead of cp1252 with a registry hack.

tsteven4, 2023-05-15:
ok.  writer fixes about ready, but you also need an option to use with garmin to force the charset.

tsteven4, 2023-05-15:
please pull the latest commits to the branch, rebuild and try
./gpsbabel -vs -w -i garmin,codec=windows-1251 -f usb: -o gpx -F test1.gpx

This should also support encoding on write (using -o garmin,codec=windows-1251)

&#x1F91E;

svintuss, 2023-05-15:
That worked, cool!

```
  <wpt lat="57.142150495" lon="33.117158515">
    <name>5 КАЛИНИНА</name>
    <cmt>5 КАЛИНИНА</cmt>
    <desc>5 КАЛИНИНА</desc>
    <sym>Lodging</sym>
  </wpt>
```

svintuss, 2023-05-15:
Sending data works too.

svintuss, 2023-05-15:
I'm a bit confused: sending .gpx works fine with this branch. But I can't send tracks from .kml, only waypoints. This happens either with or without `codec=windows-1251` option. Main branch ver. 1.8.0 sends .kml tracks correctly (apart from cyrillic).

tsteven4, 2023-05-15:
The garmin reader and writers don't know where the data comes from or goes to. The readers all convert the recovered 8 bit data, usually assuming it is UTF-8, to UTF-16, and the writers all convert this UTF-16 data to something, usually UTF-8.  The garmin codec option sets that conversion to/from UTF-16 to use whatever codec you asked for.  Note you do need to specificy the codec option with -i garmin,codec= AND -o garmin,codec=, or you will get the default.

There where multiple changes to the garmin module that independently effect waypts, routes and tracks.

What exactly do you mean by "I can't send tracks"?  Is there an error message?  crash?  Is the data not send to the device?  Or is the wrong data sent do the device?

tsteven4, 2023-05-15:
I do see something I missed.  It might cause the waypoints in routes to be mis-encoded when writing to the device, just like 1.8.0.

svintuss, 2023-05-15:
Sorry for not being specific enough. No `.kml` tracks are sent to device with `garmincodec` branch. 
If I checkout the `main` branch and recompile gpsbabel `.kml` tracks are sent to device just fine. 

No errors or log data is written to output with either branch.

tsteven4, 2023-05-15:
Please add the "-D 2" option and look for differences.  That should enable further debug output, but not having a device I am not sure what you will see.

Perhaps we are sending an illegal character to the device.

svintuss, 2023-05-15:
BTW what is the list of possible codecs? 
And is it possible to add a `translit` codec which simply converts Cyrillic letters to its Latin analogues (this is one example of transliteration tool: https://translit.cc)?

tsteven4, 2023-05-15:
You can get the list of codecs through the back door by:
gpsbabel -i unicsv,codec=xxx -f /dev/null

tsteven4, 2023-05-15:
On MY system (ubuntu jammy, with ICU):
Available Codecs:
MIBenum Name                    Aliases
3       US-ASCII                iso-ir-6, ANSI_X3.4-1968, ANSI_X3.4-1986, ISO_646.irv:1991, ISO646-US, us, IBM367, cp367, csASCII
4       ISO-8859-1              latin1, CP819, IBM819, iso-ir-100, csISOLatin1
5       ISO-8859-2              ibm-912_P100-1995, ibm-912, ISO_8859-2:1987, latin2, csISOLatin2, iso-ir-101, l2, 8859_2, cp912, 912, windows-28592
6       ISO-8859-3              ibm-913_P100-2000, ibm-913, ISO_8859-3:1988, latin3, csISOLatin3, iso-ir-109, l3, 8859_3, cp913, 913, windows-28593
7       ISO-8859-4              ibm-914_P100-1995, ibm-914, latin4, csISOLatin4, iso-ir-110, ISO_8859-4:1988, l4, 8859_4, cp914, 914, windows-28594
8       ISO-8859-5              ibm-915_P100-1995, ibm-915, cyrillic, csISOLatinCyrillic, iso-ir-144, ISO_8859-5:1988, 8859_5, cp915, 915, windows-28595
9       ISO-8859-6              ibm-1089_P100-1995, ibm-1089, arabic, csISOLatinArabic, iso-ir-127, ISO_8859-6:1987, ECMA-114, ASMO-708, 8859_6, cp1089, 1089, windows-28596, ISO-8859-6-I, ISO-8859-6-E, x-ISO-8859-6S
10      ISO-8859-7              ibm-9005_X110-2007, ibm-9005, 8859_7, greek, greek8, ELOT_928, ECMA-118, csISOLatinGreek, iso-ir-126, ISO_8859-7:1987, windows-28597, sun_eu_greek
11      ISO-8859-8              ibm-5012_P100-1999, ibm-5012, hebrew, csISOLatinHebrew, iso-ir-138, ISO_8859-8:1988, ISO-8859-8-I, ISO-8859-8-E, 8859_8, windows-28598, hebrew8
12      ISO-8859-9              ibm-920_P100-1995, ibm-920, latin5, csISOLatin5, iso-ir-148, ISO_8859-9:1989, l5, 8859_9, cp920, 920, windows-28599, ECMA-128, turkish8, turkish
13      ISO-8859-10             iso-8859_10-1998, iso-ir-157, l6, ISO_8859-10:1992, csISOLatin6, latin6
16      ISO-2022-JP-1           ISO_2022,locale=ja,version=1, JIS_Encoding, csJISEncoding, ibm-5054, JIS, x-windows-50221
17      Shift_JIS               ibm-943_P15A-2003, ibm-943, MS_Kanji, csShiftJIS, windows-31j, csWindows31J, x-sjis, x-ms-cp932, cp932, windows-932, cp943c, IBM-943C, ms932, pck, sjis, ibm-943_VSUB_VPUA, x-MS932_0213, x-JISAutoDetect
18      EUC-JP                  euc-jp-2007, Extended_UNIX_Code_Packed_Format_for_Japanese, csEUCPkdFmtJapanese, X-EUC-JP, eucjis, ujis
30      US-ASCII                iso-ir-6, ANSI_X3.4-1968, ANSI_X3.4-1986, ISO_646.irv:1991, ISO646-US, us, IBM367, cp367, csASCII
36      windows-949             windows-949-2000, KS_C_5601-1987, KS_C_5601-1989, KSC_5601, csKSC56011987, korean, iso-ir-149, ms949, x-KSC5601
37      ISO-2022-KR             ISO_2022,locale=ko,version=0, csISO2022KR
38      windows-949             windows-949-2000, KS_C_5601-1987, KS_C_5601-1989, KSC_5601, csKSC56011987, korean, iso-ir-149, ms949, x-KSC5601
39      ISO-2022-JP             ISO_2022,locale=ja,version=0, csISO2022JP, x-windows-iso2022jp, x-windows-50220
40      ISO-2022-JP-2           ISO_2022,locale=ja,version=2, csISO2022JP2
57      GBK                     windows-936-2000, CP936, MS936, windows-936
81      ISO-8859-6              ibm-1089_P100-1995, ibm-1089, arabic, csISOLatinArabic, iso-ir-127, ISO_8859-6:1987, ECMA-114, ASMO-708, 8859_6, cp1089, 1089, windows-28596, ISO-8859-6-I, ISO-8859-6-E, x-ISO-8859-6S
82      ISO-8859-6              ibm-1089_P100-1995, ibm-1089, arabic, csISOLatinArabic, iso-ir-127, ISO_8859-6:1987, ECMA-114, ASMO-708, 8859_6, cp1089, 1089, windows-28596, ISO-8859-6-I, ISO-8859-6-E, x-ISO-8859-6S
84      ISO-8859-8              ibm-5012_P100-1999, ibm-5012, hebrew, csISOLatinHebrew, iso-ir-138, ISO_8859-8:1988, ISO-8859-8-I, ISO-8859-8-E, 8859_8, windows-28598, hebrew8
85      ISO-8859-8              ibm-5012_P100-1999, ibm-5012, hebrew, csISOLatinHebrew, iso-ir-138, ISO_8859-8:1988, ISO-8859-8-I, ISO-8859-8-E, 8859_8, windows-28598, hebrew8
104     ISO-2022-CN             ISO_2022,locale=zh,version=0, csISO2022CN, x-ISO-2022-CN-GB
105     ISO-2022-CN-EXT         ISO_2022,locale=zh,version=1
106     UTF-8
109     ISO-8859-13             ibm-921_P100-1995, ibm-921, 8859_13, windows-28603, cp921, 921, x-IBM921
110     ISO-8859-14             iso-8859_14-1998, iso-ir-199, ISO_8859-14:1998, latin8, iso-celtic, l8
111     ISO-8859-15             ibm-923_P100-1998, ibm-923, Latin-9, l9, 8859_15, latin0, csisolatin0, csisolatin9, iso8859_15_fdis, cp923, 923, windows-28605
113     GBK                     windows-936-2000, CP936, MS936, windows-936
114     GB18030                 gb18030, ibm-1392, windows-54936
1000    UTF-16
1001    UTF-32
1011    SCSU                    ibm-1212, ibm-1213
1012    UTF-7                   windows-65000, unicode-1-1-utf-7, unicode-2-0-utf-7
1013    UTF-16BE
1014    UTF-16LE
1015    UTF-16
1016    CESU-8                  ibm-9400
1017    UTF-32
1018    UTF-32BE
1019    UTF-32LE
1020    BOCU-1                  csBOCU-1, ibm-1214, ibm-1215
2004    hp-roman8               ibm-1051_P100-1995, ibm-1051, roman8, r8, csHPRoman8, x-roman8
2005    Adobe-Standard-Encoding ibm-1276_P100-1995, ibm-1276, csAdobeStandardEncoding
2009    IBM850                  ibm-850_P100-1995, ibm-850, cp850, 850, csPC850Multilingual, windows-850
2013    IBM862                  ibm-862_P100-1995, ibm-862, cp862, 862, csPC862LatinHebrew, DOS-862, windows-862
2016    IBM-Thai                ibm-838_P100-1995, ibm-838, IBM838, csIBMThai, cp838, 838, ibm-9030
[... cut ...]
2253    windows-1253            ibm-5349_P100-1998, ibm-5349, cp1253
2254    windows-1254            ibm-5350_P100-1998, ibm-5350, cp1254
2255    windows-1255            ibm-9447_P100-2002, ibm-9447, cp1255
2256    windows-1256            ibm-9448_X100-2005, ibm-9448, cp1256, x-windows-1256S
2257    windows-1257            ibm-9449_P100-2002, ibm-9449, cp1257
2258    windows-1258            ibm-5354_P100-1998, ibm-5354, cp1258
2259    TIS-620                 windows-874-2000, windows-874, MS874, x-windows-874
2107    TSCII

svintuss, 2023-05-15:
> Please add the "-D 2" option and look for differences. That should enable further debug output, but not having a device I am not sure what you will see.
> 
> Perhaps we are sending an illegal character to the device.

I've found out the culprit. `garmincodec` branch works fine, it's just `gpsbabel` won't send both waypoints and tracks from `.kml` at the same time. E. g. 
`% ./gpsbabel -vs -wt -i kml -f /Users/alex/Desktop/Северка.kml -o garmin,codec=windows-1251 -F usb:` 
didn't transfer tracks whereas
`% ./gpsbabel -vs -t -i kml -f /Users/alex/Desktop/Северка.kml -o garmin,codec=windows-1251 -F usb:`
worked OK.

tsteven4, 2023-05-15:
OK.  I still missed encoding of waypoint names in routes when writing to the device.

tsteven4, 2023-05-15:
It looks like this existing code, which so far is unchanged with #1117,
https://github.com/GPSBabel/gpsbabel/blob/488aa41264c3c94c8c6e1d3de41ebd0c54d0c0d9/garmin.cc#L1012-L1021
can write past the end of rte->ident if the shortname has sufficient valid characters.

tsteven4, 2023-05-15:
> It looks like this existing code, which so far is unchanged with #1117,
> 
> https://github.com/GPSBabel/gpsbabel/blob/488aa41264c3c94c8c6e1d3de41ebd0c54d0c0d9/garmin.cc#L1012-L1021
> 
> 
> can write past the end of rte->ident if the shortname has sufficient valid characters.

I assert this is fixed in https://github.com/GPSBabel/gpsbabel/pull/1117/commits/9bea843e08c699d427c34351cd66ca76d5acf819

Please pull the latest changes, rebuild and test writing a track.  please build with "-DCMAKE_BUILD_TYPE=Debug".  This will enable assertions, and we want to make sure none fail.

For some historical reason the name of waypoints that make up a route/track is restricted to a very limited set of characters, "ABCDEFGHIJKLMNOPQRSTUVWXYZ0123456789 ", and maybe "+-", and maybe the lower case equivalents.  Other characters are thrown out.

GPSBabelDeveloper, 2023-05-16:
> I assert this is fixed in 9bea843

Honestly, that reason is cowardice.

The original Garmin spec said uppercase characters and digits only. Some
devices allowed you to do more than that and some devices would crash at
upload time or, worse, in the field if you sent characters outside that
space.

At the time the 60C's hit, Garmin refused to publish the specs and USB was
my day job so I reverse engineered the protocol, happily finding it was
mostly their serial protocol smuggled inside a (poorly implemented) USB
protocol. After they'd shipped a couple of years, Garmin eventually
published a spec for their windows USB driver which revealed some secrets
and then, later, an actual USB protocol spec and we then went back and
filled in our missing understandings.

Geocachers were a HUGE percentage of our users at the time and they only
really cared about waypoints. I could push a release to that crowd
(especially via GSAK) and get meaningful testing on dozens of models within
a few days. We discovered which boundaries we could push and which we
couldn't. The route code and track code (especially writing tracks) were
relatively unused.

It's entirely probable that the units that take CP1252 for waypoitns will
take them for routes and tracks, too. If he's in a good position to test
this, go for it.

Thanx for poking at the history.

svintuss, 2023-05-16:
> Please pull the latest changes, rebuild and test writing a track. please build with "-DCMAKE_BUILD_TYPE=Debug". This will enable assertions, and we want to make sure none fail.

I have successfully built the latest `garmincodec` commit with `-DCMAKE_BUILD_TYPE=Debug`. Track sending works fine, their names are properly converted to cp1251. 

TBH I don't quite understand what exactly should I test with track/route waypoints. Maybe I just don't have tracks with custom point names. Am I getting it right that track waypoint name resides in `<ele>` `.gpx` tag?
```
      <trkpt lat="69.444419732" lon="30.763811963">
        <ele>164.042</ele>
      </trkpt>
```

svintuss, 2023-05-16:
> If "my ride" has non-ASCII characters (i.e. "my řȋɗê" or whatever is actually representable in 1252) does it now successfully roundtrip to/from the receiver?

Thank you for explanation. 
My GPSmap 60Cx handles non-ASCII track and route names correctly. And now I can send them to and from device.

tsteven4, 2023-05-16:
There were multiple other places we could have overwritten data structures that also have been fixed.  Each of these 4 memcpy calls used the size of the source, and in some cases, although unlikely, it could have been larger than the destination.  All of these were fixed previously so @svintuss has already ran this code.
https://github.com/GPSBabel/gpsbabel/blob/488aa41264c3c94c8c6e1d3de41ebd0c54d0c0d9/garmin.cc#L879-L900

tsteven4, 2023-05-16:
The waypoints that compose our tracks and routes can have names.  Reading a Garmin eTrexLegend manual, their route are composed of waypoints that can have names.  The suspect cleaning of all but A-Z, 0-9, space and perhaps + and - is related to the names of waypoints that make up a route or a track.  Perhaps the waypoint names of the waypoints that make up a track get dropped somewhere in garmin land, but it does seem like the waypoint names of waypoints that make up a route should exist in garmin land.  But I think we butcher them with the suspect cleaning mentioned above.

@svintuss, can you read this gpx file and write the route to your device, and then look at the names of the points that make up the route?  The points that make up the route are name RPT001, rpt005 and RPT011.  I think your device is recognized with id=292, which won't force uppercase.  Therefore I think on your device you will see name RPT001, 005, and RPT011.  I think if you rename these in the gpx file to Cyrillic characters they will all be filtered out and the names on your device will be empty (unless garmin makes something up). 

[testrte.gpx.txt](https://github.com/GPSBabel/gpsbabel/files/11491591/testrte.gpx.txt)

svintuss, 2023-05-17:
> The waypoints that compose our tracks and routes can have names. Reading a Garmin eTrexLegend manual, their route are composed of waypoints that can have names. The suspect cleaning of all but A-Z, [...]

Hi. 
I've uploaded `testre.gpx` to my GPSmap 60cx with `gpsbabel`. Route points were indeed converted to uppercase and any Cyrillic characters I've manually added (like `<name>Тчк 3</name>`) were truncated to ASCII and numerals.
However I tested that same `.gpx` with Cyrrillic points against MapSource on a Win7 machine and it uploaded both upper- and lowercase as well as Cyrillic symbols in route waypoint names without issue. 

So I think we should be more flexible with point names limitations: GPSmap 60Cx can display cp1251 and cp1252 in route and track waypoints.

tsteven4, 2023-05-17:
@robertlipe
1. It doesn't fix it.  See below.
2. There are bigger problems as reported https://github.com/GPSBabel/gpsbabel/issues/1115#issuecomment-1551017591

This test, cut from the latest garmin.cc, outputs "ABC123", the lower case letters abc are dropped.

```
#define MILITANT_VALID_WAYPT_CHARS "ABCDEFGHIJKLMNOPQRSTUVWXYZ0123456789"

/* Technically, even this is a little loose as spaces arent allowed */
static const char* valid_waypt_chars = MILITANT_VALID_WAYPT_CHARS " ";
static QRegularExpression invalid_char_re;

int main()
{
  assert(!QString(valid_waypt_chars).contains('\\'));
  assert(!QString(valid_waypt_chars).contains('^'));
  assert(!QString(valid_waypt_chars).contains('-'));
  assert(!QString(valid_waypt_chars).contains('['));
  assert(!QString(valid_waypt_chars).contains(']'));
  invalid_char_re = QRegularExpression(QStringLiteral("[^%1]").arg(valid_waypt_chars));
  assert(invalid_char_re.isValid());

  QString cleanname = "abcABC123";
  bool receiver_must_upper = false;

  if (receiver_must_upper) {
    cleanname = cleanname.toUpper();
  }
  cleanname = cleanname.remove(invalid_char_re);
  qDebug() << cleanname;
}
```

tsteven4, 2023-05-17:
@robertlipe 
Of course this will fix the lower case cleaning issue.  But we also use valid_waypt_chars with shortname, so it is a bit more involved.
```
invalid_char_re = QRegularExpression("[^a-zA-Z0-9 +-]");
```

`"abc +-ABC123*&@!"  ->  "abc +-ABC123"`

But why are we cleaning the names of waypoints in routes or tracks this way, while we clean the names of isolated waypoints a different way (mkshort)?   I am cautious of changes here as they will affect all the different garmin units, but it is a little hard to believe the requirements on the names of isolated waypoints are different than the requirements on the names of waypoints in routes.

tsteven4, 2023-05-17:
@robertlipe 
The difference in name cleaning came in here https://github.com/GPSBabel/gpsbabel/commit/de992ba4291f597e5d44db850205cc1cd8287c96

Previously we used mkshort on isolated waypoint names, and nothing on route/track waypoint names.

tsteven4, 2023-05-17:
> but it is a little hard to believe the requirements on the names of isolated waypoints are different than the requirements on the names of waypoints in routes.

In looking at various versions of the Garmin device interface spec I see they did have different requirements for "User Waypoint Idendifiters" (upper-case letters, numbers) and "Route Waypoint Identifiers" (any ASCII character).  They also say the host should be able to receive any ASCII character, but only send the restricted set to the GPS.  And the say some products may allow additional characters, but no attempt is made to document these product specific additions.   &#9760;

tsteven4, 2023-05-17:
Maybe we follow this lead with rtept names, only cleaning them if receiver_must_upper.  This would work for @svintuss, the question is do we break anybody other devices that set receiver_must_upper to 0.
https://github.com/GPSBabel/gpsbabel/blob/4aa594a348231eb90ccb20bd7a140a949786bc7a/garmin.cc#L330-L338

tsteven4, 2023-05-17:
Which would gets us to the following, sidestepping the issue of cleaning of a-z issue.
```
  QString cleanname = wpt->shortname;
  if (receiver_must_upper) {
    cleanname = cleanname.toUpper().remove(invalid_char_re);
  }
  write_char_string(rte->ident,
                    str_from_unicode(cleanname).constData(),
                    sizeof(rte->ident));
```

svintuss, 2023-05-18:
I've made a `git pull` and recompiled `garmincodec` branch. However seems like I've already built the latest commit last time:
```
% cmake --build .
ninja: no work to do.
```

Nevertheless I reuploaded `testrte.gpx` with the following data to GPSmap 60cx:
```
<name>Road Дорога</name>
<desc>Generated from track Route</desc>
<rtept lat="36.339560000" lon="-117.422570000">
  <ele>479.3</ele>
  <name>Point 1</name>
</rtept>
<rtept lat="36.345010000" lon="-117.354270000">
  <ele>624.3</ele>
  <name>ТОЧКА 2</name>
</rtept>
<rtept lat="36.344640000" lon="-117.349030000">
  <ele>663.7</ele>
  <name>_~-$#``;""|\/ ABC321</name>
</rtept>
```

Gpsbabel got me `Road Дорога` route name and `P 1`, ` 2` and ` ABC321` route points.
Mapsource got me `Road Дорога` route name and `Point 1`, `ТОЧКА 2` and `_~-$#``;""|\/ ABC321` route point names on the unit.

May I suggest a manual switch to enable/disable point names truncation?
And maybe a transliteration function to upload Cyrillic letters on old units with original Latin1 firmware?

tsteven4, 2023-05-18:
@svintuss How did you do this to your GPSmap 60cx: "Extended latin font characters were replaced with Cyrillic counterparts"

We are chewing over how to proceed with the rtept names.

I don't think we know how to upload alternate character sets to the device.

svintuss, 2023-05-18:
I've installed a custom firmware like the ones hosted here: [](http://e-trex.narod.ru/download_en.htm)
As I understand it back in the day it was the only way to implement Cyrillic support. Here is a link to `flawia` project, which hosts a font editor tool: [](https://sourceforge.net/projects/flawia/files/fe/1.0/)

Basically there are units with unmodified firmware where users tend to upload maps and use waypoints in Latin1 with transliteration (i. e. `Я -> JA, Ч -> CH`) and the ones with alternative/modified firmware which accept cp1251 maps and points.

tsteven4, 2023-05-18:
Thanks.  The hosted here part and the links didn't come through, but I get the idea.  ICU has such a [transform](https://unicode-org.github.io/icu/userguide/transforms/general/#scriptlanguage), but we don't always have ICU available and I think it is unlikely we would invest the development effort in this day of unicode dominance.

However, I am hopefully that we can stop trashing the rtept names when writing to the device.  We know how to do this, but we want to be careful because many devices are supported and they may not all tolerate unexpected characters.  One choice would be to throw it onto the user with an option, but we already are making this decision for users on some devices, like yours, when sending other fields besides the rtept names.

svintuss, 2023-05-18:
I'll fix a GPS V in a couple of days and could check whether it supports non-alfanumeric and lower-case letters in route points.
As for GPSmap 60cx, it definetely can handle non-truncated rtepts.

> Thanks. The hosted here part and the links didn't come through, but I get the idea. ICU has such a [transform](https://unicode-org.github.io/icu/userguide/transforms/general/#scriptlanguage), but we [...]

It leaves us with a unix-way approach: pipe the gpx output through an external converter which will replace Cyrillic with translit.

closed 2023-05-19 by tsteven4

svintuss, 2023-05-19:
> So all this is about/for a custom, modified, unofficial firmware version? I don't love the nested levels of "unsupported" in all this. 

First of all I'm very grateful to you guys for implementing encoding conversion. It really makes sending and receiving POIs much simpler without the need of post-processing gpx'es.

Secondly I do agree that the devices I mention are indeed obsolete, but they are still useful and functional especially for those of us who don't want to produce more e-waste. 
As of levels of "unsupported" I can't completely agree. Back in the day there were officially translated Garmin units (e. g. eTrex H series) but they were poorly translated and generally buggy and the market went for hacking Latin1 units. So although my device has unofficial firmware installed there is nothing unsupported in cp1251 units in general. After all MapSource sends cp1251 maps, POIs, routes and tracks without issue.

tsteven4, 2023-05-19:
We needed to do this work anyway to get cp1252 fixed which we use for some devices by default.

In any event, the rtept names should work on your 60cx now if you build from the master branch or use our pre-release installer https://github.com/GPSBabel/gpsbabel/releases/tag/Continuous-macOS

Thanks for the debugging.  It definitely wouldn't have happened without your help.

PR #1117 (2023-05-19): fix garmin serial/usb device character encoding/decoding.
We got away with breaking this for a long time, but we finally got caught.

This resolves #1115

@svintuss has tested both read and write on his Garmin GPSmap 60cx using windows-1251.