GPSBabel/gpsbabel #1123: filter_track.html corrections

jidanni, 2023-06-04:
[filter_track.html](https://www.gpsbabel.org/htmldoc-development/filter_track.html) says:

> Split by date or time interval (see README).

https://www.gpsbabel.org/readme.html ? Probably not. How about one of these?

```
$ dlocate -i gpsbabel.*REA|sed s/.*\\///
README.Debian
README.contrib.gz
README.igc.gz
README.mapconverter
README.md
```
Nope.

So please remove old reference to README or hyperlink it if still relevant.

> gpsbabel -t -i gpx -f in.gpx -x track,pack,split,title="ACTIVE LOG # %D" -o gpx -F out.gpx

Be sure to mention somewhere on the page that these (I hope) are strptime formats.

> gpsbabel -t -i gpx -f in.gpx -x track,pack,sdistance=0.1k" -o gpx -F out.gpx

Remove unmatched quote.

> segment option

> segment tracks with abnormally long gaps.

1 day? 1 hour? 1 mile? Mention how defined.

tsteven4, 2023-06-04:
see README &#9989;
unmatched quote &#9989;
strptime formats &#10060;  The title option mentions strftime already
segment &#10067; This was added in 36ff778d6.  The answer the points need to be greater than 0.001 radian = 6378m apart and greater than 1.2 times a running average of the distance between points.  The option is all hard coded for some specific purpose that I don't know.  It is untested.  @robertlipe should we even keep this option?

GPSBabelDeveloper, 2023-06-04:
Yes. Earth uses this to deblobify static/slow moving tracks without
actually using the track filter.

It's admittedly not great.