GPSBabel/gpsbabel #1130: ‘kVersionDate’ was not declared in this scope

PacketFiend, 2023-06-15:
Compilation now fails for me, with the following error:
```
gpsbabel/gui/mainwindow.cc:1151:42: error: ‘kVersionDate’ was not declared in this scope; did you mean ‘kVersionSHA’?
 1151 |   QDateTime date = QDateTime::fromString(kVersionDate, Qt::ISODate);
      |                                          ^~~~~~~~~~~~
      |                                          kVersionSHA
gmake[2]: *** [gui/CMakeFiles/gpsbabelfe.dir/build.make:332: gui/CMakeFiles/gpsbabelfe.dir/mainwindow.cc.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:287: gui/CMakeFiles/gpsbabelfe.dir/all] Error 2
gmake: *** [Makefile:101: all] Error 2
```

`git blame` is telling me to ask @tsteven4:
```
de86a9869 gui/mainwindow.cc           (tsteven4 2023-04-26 10:21:59 -0700 1151)   QDateTime date = QDateTime::fromString(kVersionDate, Qt::ISODate);
de86a9869 gui/mainwindow.cc           (tsteven4 2023-04-26 10:21:59 -0700 1152)   QString utcdate;
de86a9869 gui/mainwindow.cc           (tsteven4 2023-04-26 10:21:59 -0700 1153)   if (date.isValid()) {
de86a9869 gui/mainwindow.cc           (tsteven4 2023-04-26 10:21:59 -0700 1154)     utcdate = date.toUTC().toString(Qt::ISODate);
de86a9869 gui/mainwindow.cc           (tsteven4 2023-04-26 10:21:59 -0700 1155)   }
de86a9869 gui/mainwindow.cc           (tsteven4 2023-04-26 10:21:59 -0700 1156)   AboutDlg aboutDlg(nullptr, babelVersion_, QString(appName) + QString(" " VERSION), kVersionSHA, utcdate, babelData_.installationUuid_);
```

The gpsbabel binary is still built, but the build fails when compiling the GUI, so, not a complete blocker for me.

tsteven4, 2023-06-15:
This builds clean for me (from  https://github.com/PacketFiend/gpsbabel.git), and all this works in regression.  You shouldn't have to fool with kVersionDate at all.   But, gbversion.h and gui/gbversion.h are created when you run the configure step of cmake.  So cmake doesn't detect that it needs to regenerate the gbversion.h files, but you do.

so rerun the cmake generate step:
[Generate a Project Buildsystem](https://cmake.org/cmake/help/latest/manual/cmake.1.html#generate-a-project-buildsystem)
 cmake [<options>] -B <path-to-build> [-S <path-to-source>]
 cmake [<options>] <path-to-source | path-to-existing-build>

```
commit 84421454586fe91c13c81483166039e0d9f46063 (HEAD -> master, origin/master, origin/HEAD)
Merge: 9afa3a52 043fe4f0
Author: PacketFiend <kenneth@voort.ca>
Date:   Thu Jun 15 12:18:10 2023 -0400

    Merge remote-tracking branch 'gpsbabel/master'
```

PacketFiend, 2023-06-15:
Thanks, rookie mistake right there.

closed 2023-06-15 by PacketFiend