GPSBabel/gpsbabel #1159: :error:fetch Failed to fetch curl-ca-bundle: The requested URL returned error: 404 Not Found

ishandutta2007, 2023-08-15:
```sudo port install gpsbabel```

Error:

```

:notice:fetch --->  Attempting to fetch curl-7.72.0.tar.xz from http://nue.de.distfiles.macports.org/curl
:debug:fetch Fetching distfile failed: The requested URL returned error: 404 Not Found
:notice:fetch --->  Attempting to fetch curl-7.72.0.tar.xz from http://aarnet.au.distfiles.macports.org/pub/macports/distfiles/curl
:debug:fetch Fetching distfile failed: The requested URL returned error: 404 Not Found
:error:fetch Failed to fetch curl-ca-bundle: The requested URL returned error: 404 Not Found
:debug:fetch Error code: NONE
:debug:fetch Backtrace: The requested URL returned error: 404 Not Found
:debug:fetch     while executing
:debug:fetch "error $lastError"
:debug:fetch     (procedure "portfetch::fetchfiles" line 66)
:debug:fetch     invoked from within
:debug:fetch "portfetch::fetchfiles"
:debug:fetch     (procedure "portfetch::fetch_main" line 17)
:debug:fetch     invoked from within
:debug:fetch "$procedure $targetname"
:error:fetch See /opt/local/var/macports/logs/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_net_curl/curl-ca-bundle/main.log for details.
```

ishandutta2007, 2023-08-15:
@robertlipe that file doesn't even exist, it's not a network issue.

ishandutta2007, 2023-08-15:
@robertlipe I need for macos 10.13. The version GPSBabel1.8.0.dmg supports macos 10.14+.

GPSBabelDeveloper, 2023-08-15:
Your error says you can't fetch curl from somewhere in Aus. That's not us
issuing or serving that.

Your is may be so old it's just not supportable any longer. The wheels on
MacOS turn pretty relentlessly.

ishandutta2007, 2023-08-15:
@GPSBabelDeveloper How to install GPSBabel on macos 10.13 (High Sierra)?

ishandutta2007, 2023-08-15:
@GPSBabelDeveloper I do understand that you are in no way responsible for file hosted by macports, and I do I understand that it may not be an issue of this repo. I am just looking for solution for my problem, and you guys seem to be the closest on who can help.  🥇