GPSBabel/gpsbabel #1182: garmin_fit does not support course_point types

googol42, 2023-09-17:
Hello,

I am using gpsbabel to convert gpx files to fit. After some debugging I have found out that gpsbabel does not take the icon types into account when converting the gpx to fit.

I tried the following combinations in my gpx file:
- `sym` tag with numeric type (e.g. `<sym>3</sym>` for water)
- `sym` tag with lower case type  (e.g. `<sym>water</sym>`)
- `sym` tag with upper case type  (e.g. `<sym>WATER</sym>`)
- `type` tag with numeric type  (e.g. `<type>3</type>`)
- `type` tag with lower case type  (e.g. `<type>water</type>`)
- `type` tag with upper case type  (e.g. `<type>WATER</type>`)

I then converted those 6 files to fit using `gpsbabel -i gpx -f file_x.gpx -o garmin_fit -F file_x.fit`

I then converted the fit files to csv using the FitCSVTool from the Fit/Garmin SDK (https://developer.garmin.com/fit/download/):

`java -jar Downloads/FitSDKRelease_21.115.00/java/FitCSVTool.jar -b file_x.fit file_x.csv`

Then I was sure that it must be an error in gpsbabel. I search your repository and found this line: https://github.com/GPSBabel/gpsbabel/blob/dfc770ac0a68a0827f89ddf8ef09c5766f16a53c/jeeps/gpscom.cc#L1282C5-L1282C34 I am not sure if this code is really used when converting....

I did not find any upstream repository for "jeeps" so I am reporting this here.

I would expect that when I convert a gpx file with a waypoint with `<type>WATER</type>` (or maybe `<type>water</type>`) to be converted to a `course_point` with `type = 3`

googol42, 2023-09-17:
Also the list of types is not complete (I know it is only a comment): https://github.com/GPSBabel/gpsbabel/blob/dfc770ac0a68a0827f89ddf8ef09c5766f16a53c/jeeps/gps.h#L224

A full list of types can be found in the mentioned SDK. Open `FitSDKRelease_21.115.00/py/garmin_fit_sdk/profile.py` and search for `'course_point':` which gives you a list of 54 types.

googol42, 2023-09-17:
I figured out that the code is not called and found this: https://github.com/GPSBabel/gpsbabel/blob/dfc770ac0a68a0827f89ddf8ef09c5766f16a53c/garmin_fit.cc#L1277

Unfortunately the intermediate "format" `Waypoint` does not support something like a type or symbol. The code above attempts (to some extend) to extract the type from the (short)name.

Could that be enhanced? For example: gpx waypoint has the name "water spring" or "WATER spring" the course_point is assigned type 3 and the name "spring". Or do you see any other way to around that limitation?

tsteven4, 2023-10-09:
> Unfortunately the intermediate "format" `Waypoint` does not support something like a type or symbol.

Actually, the gpx reader records the value of the sym element as the field icon_descr in the Waypoint class.

GPSBabelDeveloper, 2023-10-10:
What he said.

We have dozens of examples (some good, some bad) of code turning numeric
values into textual strings inside icon_descr and then doing the inverse
for the opposite in read/write.  Supporting round-trip to GPX/CSV is table
stakes.  You get bonus (imaginary internet) points for applying the Postel
Principle. If you can take take either "red flag" and "flag, red" on read
and convert those to 1,475 internally (or whatever) and then write the
"best" on output, everyone wins.

hummin_bird_icons is a reasonable example.
osm_icon_mapping is a featureful example as they have complicated problems
garmin_icon_tables contains hundreds of icon # mappings as used in their
receiver and mapping products. It's complicated because "Airport" is 107 in
Mapsource and 16384 in PCX, for example.

It looks like the code you've pointed to kind of tries to do this, but only
for five languages worth of "left" and 'right". That is, well, interesting.

Our older formats tend to use brute force traditional linear table lookups.
Newer code tends to use more C++  indigenous structures like maps and
hashes

If you have knowledge of a number<->string mapping for icons, there are
plenty of examples for you to add here.  Looks like you found the place on
read and probably let fit_parse_data look in lappt (?) f.course_data_type
or whatever to look up the textual valu of that number. We can work with
you to get it integrated if you can work out the tymbol table and get it
basically working from your device/.