GPSBabel/gpsbabel #1184: www.gpsbabel.org doesn't contain a link to github page

daald, 2023-09-22:
I just browsed through the official website https://www.gpsbabel.org/, looking for the source code. There is a lot of documentation (which I appreciate), three binary downloads, and many more, but finally I had to google for the source code to find this github project. If the link is somewhere, it is very very hidden. Why?

I would like to see a link to the github project somewhere on the first page.

daald, 2023-09-22:
[THIS](https://www.gpsbabel.org/download.html) is the download page.
And somewhere in the paragraph "GPSBabel is free software" of the main page would also be helpful.

I'm just telling my user experience. It took me 5 minutes what should be 20 seconds.

GPSBabelDeveloper, 2023-09-22:
[image: Screenshot 2023-09-22 at 6.10.11 PM.png]