GPSBabel/gpsbabel #1195: gpx.h GEOTAG searches for elements that imply a gpx schema violation

tsteven4, 2023-10-24:
Back in 2007 a couple of commits attempted to add different ways to find geocaches in the gpx reader:

1. e8cc8f6f2 12/6/2007 /gpx/wpt/geocache added 'Ralf Dragon adds support for opencaching.de GPX reads'
2. 0ac4917ea 2/25/2007 /gpx/wpt/extensions/cache added 'Allow groundspeak namespace in GPX 1.1 or 1.0 format'

Neither of these has a test case.

The current definition of GEOTAG is https://github.com/GPSBabel/gpsbabel/blob/1aa1554c03158d42346ae7adf600ede41a382a4f/gpx.h#L328-L331

`{"/gpx/wpt/extensions/cache/" name, {type, true}}, \`
implies a schema violation in either gpx 1.0 (extensions element not expected) or gpx 1.1 (cache element not expected).

`{"/gpx/wpt/geocache/" name, {type, true}}     /* opencaching.de */`
implies a scheam violation of either gpx 1.0 (geocache element not expected) or gpx 1.1 (geocache element not expected).

The gpx 1.0 and 1.1 schema uses
`<xsd:any namespace="##other" ...`
The issue is the ##other restriction is violated.

Today opencaching.de uses gpx 1.0 with groundspeak:cache and oc:cache elements (`xmlns:oc="https://github.com/opencaching/gpx-extension-v1"`)
I don't have any recent examples of caches from geocaching.com so it is unclear what they are doing now.
I don't know that any creator ever used the groundspeak namespace in gpx 1.1.

a test script, log and reference files:
[test_files.zip](https://github.com/GPSBabel/gpsbabel/files/13114471/test_files.zip)

tsteven4, 2023-10-24:
I was able to download a few recent gpx caches from opencaching.de without agreeing to anything.  We may be able to use those as reference files.

If you can get a few current gpx caches from groundspeak that we can use as references that would be very good.

tsteven4, 2023-10-24:
> It might be nice if we can snag difficulty, terrain, etc. out of an opencaching.de file so we can convert it to unicsv

Without any changes:
```
gpsbabel -i gpx -f OC15F8D_10.gpx -o unicsv -F -
No,Latitude,Longitude,Name,Description,Symbol,Date,Time,URL,GCID,Type,Container,Terrain,Difficulty,Archived,Available,Last Found,Placer,Placer ID,Hint
1,50.174620,8.687800,"OC15F8D","TB Hotel Storchenwiese","Geocache",2020/04/13,18:00:00,"https://www.opencaching.de/viewcache.php?cacheid=193085",193085,"Traditional Cache","Regular",1.5,1.5,"False","True","2022/12/03 17:00:00","Bullitt25",361790,"Andere Seite, Hecke unten große Dose."
```

tsteven4, 2023-10-24:
Regarding the groundspeak extension(s), adding support for gpx 1.1 requires multiple changes to the gpx reader hash table, there are a bunch of places the groundspeak namespace prefix is missing.

Regarding opencaching.de, elements geocache and hints don't seem to be listed in any xsd they use, and don't appear in samples I looked at today.  They also say `As far as we know, all Geocaching sites export their GPX files in Topografix 1.0 format. ` on https://github.com/opencaching/gpx-extension-v1/blob/master/all-these-namespaces.md.

@robertlipe if you can get some recent vintage samples from geocaching.com it would be good to 
- verify they still use gpx 1.0
- if possible, add a few as references, perhaps replacing our ancient references (that have been manually altered).

The way this is going I think we should drop support for gpx 1.1 and geocaches.  I have my doubts that it every worked, and it doesn't work now even with valid manually converted gpx 1.1 caches.  It is interesting that the comments in xsd files often cite a particular version of gpx, but I don't see any enforcement of that in the schemas (e.g. https://github.com/opencaching/gpx-extension-v1, which we don't support, and http://www.garmin.com/xmlschemas/GpxExtensions/v3, which we do)

GPSBabelDeveloper, 2023-10-25:
As expected, you are wise.

I thought I typed a response to this earlier, but don't see it.

To my knowledge, the intersection of GPX 1.1 and Geocaching (sites that I
care about that have made themselves known to us at all) is the null set.
Geocaching.com uses GPX 1.0 and their extensions are GPX 1.0.  Because the
geocaching files are read by millions (?) of GPS receivers in the market
that'll never get a firmware update - and because of past upheavals even
when Groundspeak made 'safe' changes - they are extremely conservative to
changing those files.

It's a sucker bet to wager what someone else will do, but given how their
pages <https://www.geocaching.com/software/default.aspx> still speak highly
of Palm/OS, Blackberry, and WinCE apps and knowing the huge mess they had
on their hands in the past when they updated PQs - both when they weren't
great with compatibility and when they did all the right things, but
readers that were in device firmware couldn't cope - I just don't see them
moving to GPX 1.1 because gains them nothing and costs them plenty because
they can't ever really drop GPX 1.0. I remember they once added a new icon
type (groundspeak:type) for an event that happened once a year and because
old receivers handled it so badly they had to revert that change.

Trivia: at one point, there were three independent GPX writers at
Groundspeak: one for pocket queries (the bulk "give me a thousand" mode),
one for downloading the single page you're looking at, and one for the
mobile-centric API that's quite garden walled.

That OC link is interesting. Their view of history is, well, it's one
interpretation. :-) The Opencaching network has more traction in EU than it
does in NA, but it's interesting that we just don't hear much about them -
and we don't make their list of software, either. {shrug}

I just proposed https://github.com/GPSBabel/gpsbabel/pull/1196 (I'm not
married to the name.)

Just skimming the diffs:

It swaps available and archived. The cache is no longer available, so this
is intentional.
It moves to "groundspeak 1.0.1" This makes more of the game attributes
tri-states instead of bools. (
Groundspeak no longer relies on the implicit 'Z' that was the subject of
consternation in GPX. (and the subject of an open bugreport for us)
They no longer write sub-atomic precision for coordinates.
HTTPS is just how the web is done these days.
They've made changes to user-generated HTML in the tags (another
tempest...) to reduce the amount of colloquial/broken HTML that the viewer
(often a moderately underpowered ARM-class device) may have to parse and
view.
They're a Windows house (Seattle company....) so they have DOS carriage
returns.

So I don't think Geocaching.com GPX files have -really- changed for us
materially in a Very Long Time.

tsteven4, 2023-10-25:
Should we just replace GCGCA8.gpx with the new version, and update the 6 changed output reference files?

tsteven4, 2023-10-25:
I found a source for xsd files that may be related to issue 1 above.  I am not sure any of these schema are in use any more.  They don't appear to be used by geocaching.com.au now.  For example, a randomly selected file from https://geocaching.com.au/cache/tp13351.gpx uses gpx 1.0 and groundspeak cache 1.0.1

https://geocaching.com.au/geocache/
https://geocaching.com.au/geocache/1/0/1/geocache.xsd uses the top level element name "geocache".
https://geocaching.com.au/geocache/1/1/geocache.xsd uses the top level element name "cache".
https://geocaching.com.au/geocache/1/geocache.xsd uses the top level element name "geocache".

In any event, this doesn't change my original point about implying a schema violation.  But, the illegal geocache element without a namespace declaration does get parsed today, where a legal one with a namespace declaration does not.

GPSBabelDeveloper, 2023-10-25:
It's possible that all (most of?) these offshoot sites decided to try to
quit wagging the dog and get over their ideological objection to using
groundspeak: so that software could actually use their files.

If the proposal is to replace
#define GEOTAG(type,name) \
  {"/gpx/wpt/groundspeak:cache/groundspeak:" name, {type, true}}, \
  {"/gpx/wpt/extensions/cache/" name, {type, true}}, \
  {"/gpx/wpt/geocache/" name, {type, true}} /* opencaching.de */
with
#define GEOTAG(type,name) \
  {"/gpx/wpt/groundspeak:cache/groundspeak:" name, {type, true}}

and accept only groundspeak: extensions and thus files that are GPX 1.0
with Groundspeak Geocaching extensions 1.0.1 or 1.0.0, consider it agreed
upon. That passes testo, but we don't claim to have samples for all the
geocaching spinoffs.

PR #1198 (2023-10-27): remove gpx reader support for probelmatic extensions.
This resolves #1195.

1. groundspeak is generally used with gpx 1.0.  Our attempt to support it with gpx 1.1 required schema violations.

2. opencaching.de no longer uses the geocache element.  Our implementation required schema violations.
closed 2023-10-27 by tsteven4