GPSBabel/gpsbabel #1204: C++20?

tsteven4, 2023-11-01:
Should we raise the floor to C++20?

## Compilers in use:

Assuming we drop toolset v141,version=14.16.27023 which is the lastest toolset for visual studio 2017, and require at least the latest toolsets for visual studio 2019 which would be v142, version 14.29.30133 or any in support toolset for visual studio 2022 which would be >= v143, version 14.32.31326.  (14+5=19 on the support features page below).

Ubuntu jammy, has g++ 11.4.0 and clang 14.0.0.

The oldest Xcode we use in CI is currently 12.5.1 on macOS 11, which has apple clang 12.0.5.  We could use Xcode 13.2.1, with apple clang 13.0.0 on macOS 11.

## Compiler/Library feature support

Support is summarized here: https://en.cppreference.com/w/cpp/compiler_support#cpp20 and here https://en.cppreference.com/w/cpp/compiler_support#C.2B.2B20_library_features

## Benefits

One features I have wanted for consistency is "Default member initializers for bit-fields", although the impact on our code set is minimal.

What other features would be useful?

tsteven4, 2023-11-01:
It looks to me like the only toolset we use with complete support is MSVC!  The community edition is free.   I think the question is how much the version of gcc and clang are going to hold us back on Ubuntu jammy, and apple clang on whatever we want to set our minimum development floor to be for macOS.

tsteven4, 2023-11-01:
The important user of the jammy LTS for us is the github runners.  jammy is currently the latest Ubuntu release available on a github hosted runner.

tsteven4, 2023-11-01:
And the first failure is on macOS, with the newest version of Xcode we currently use:

Applications/Xcode_14.2.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/c++/v1/concept:318:1: error: Parse error at "::"

tsteven4, 2023-11-01:
above error with concepts is reported at https://bugreports.qt.io/browse/QTBUG-117765

tsteven4, 2023-11-01:
@robertlipe what version of macOS are you trying to keep alive for building?  As an alternative you likely can use teamviewer or ssh to our macstadium machine.  I believe more than one developer is allowed but usage is restricted to development of gpsbabel.

GPSBabelDeveloper, 2023-11-01:
Sheesh. We might not care about modules, but I'm sure that Qt cares deeply
about them. It's interesting that older versions of Qt don't trip this up.

It's also interesting how that bug and most that link to it are very
contemporary - active in recent weeks.

I'm surprised that moc has lived as long as it has, but the templated
alternatives I've seen that try to automate the signal/slot plumbing have
been even more terrifying.

tsteven4, 2023-11-01:
see PR #1206 for latest CI results

GPSBabelDeveloper, 2023-11-01:
My 27" 2012 iMac with 10.15.7 (3.4Ghz, 32GB, external 'fusion' drive) is
still a pretty likeable desktop. If I could use it as a monitor, I would
probably feed it from a Mini or something. I carry a 13" 2020 M1 MBP around
the house for most of my tinkering. I can ssh/screen share to that or my
wife's M1 iMac. Let's not hold back development because I'm using a machine
I make fun of others for using. :-).

I understand there's a way to hack more modern OSes to run on it, but it
runs into the kinds of problems that I buy Macs to avoid.

Shame on Apple for talking green but making e-waste by discontinuing OS
support so aggressively. But that's a dog I can't wag.

Let's not let that machine factor into larger project issues.

> see PR #1206 <https://github.com/GPSBabel/gpsbabel/pull/1206> for latest
CI results

It is interesting that it's one specific configuration that fails. Maybe
newest Qt and older (Sonoma was released, what, two weeks ago?) MacOS just
isn't a combination we test. It might be a bit nonsensical. I wonder what
percentage of Mac devs (and I've already acknowledged we don't have many at
all...) really run Qt installers vs. running it from homebrew which
probably patches around this kind of thing.

It looks like my M1 is running homebrew qt
 qmake --version
QMake version 3.1
Using Qt version 6.5.2 in /opt/homebrew/lib

(Probably because I was an early M1 adopter at Qt was one of the last open
source blockers)

While my x86 iMac is running a manually installed Qt version 6.4.2

tsteven4, 2024-08-19:
closed via #1206

closed 2024-08-19 by tsteven4