GPSBabel/gpsbabel #1224: Broken link to Entire Manual as PDF

jidanni, 2023-11-17:
1. On https://www.gpsbabel.org/readme.html
2. Click [Entire Manual as PDF](https://www.gpsbabel.org/htmldoc-development/gpsbabel.pdf)

tsteven4, 2023-11-19:
Yes, the links are broken.  However, the material for the latest release is on the server:
https://www.gpsbabel.org/htmldoc-1.9.0/index.html
https://www.gpsbabel.org/htmldoc-1.9.0/gpsbabel-1.9.0.pdf

tsteven4, 2023-11-20:
perhaps we can write release notes as we do commits, or flag commits worthy of mention when we commit them.

GPSBabelDeveloper, 2023-11-20:
I just added a tag for needs-release-notes.  We can apply it via the web
interface to either the issue or the PR.

Let's give that a try. Maybe that'll make creating the notes less dreadful.