GPSBabel/gpsbabel #1255: Exporting as GPX always fills "<cmt>" and "<desc>" even if not existent in the source file

M-Reimer, 2024-01-21:
I'm not quite sure if this is a bug or intended behavior or a bug.

My source file has a waypoint like this:

```
<wpt lat="12.34" lon="12.34">
  <time>2015-11-13T15:26:05Z</time>
  <name>Am Hohenrod</name>
</wpt>
```

If I just pass this through gpsbabel, it results in:

```
<wpt lat="12.34" lon="12.34">
    <time>2015-11-13T15:26:05Z</time>
    <name>Am Hohenrod</name>
    <cmt>Am Hohenrod</cmt>
    <desc>Am Hohenrod</desc>
  </wpt>
```

And this actually also happens with other export formats.

Why?

M-Reimer, 2024-01-21:
This IMHO should be optional. I want to deliver my files over the internet and this is wasted space and takes longer to transmit.

For me the much bigger problem is that I have a similar issue with GeoJSON. The Name is copied to the description there also.

GPSBabelDeveloper, 2024-01-21:
Lucky for you, compression has been the norm on the internet for most of
the last 20 years. Those additional lines with completely identical
tokens cost almost nothing.

If you control the reader and know exactly what fields will be harvested,
you can disable the code at the end of waypt_add(), but we have to support
millions of users with hundreds of applications.

M-Reimer, 2024-01-22:
So if this is required for GPX to support some special cases, would it be OK to clean that up for GeoJSON export at least?

I wanted to use GeoJSON for a web map (Leaflet) and I need to know there if there was a description defined as having a description changes how the POI looks (POIs with a description can be clicked and the description displays in a popup). If the name is duplicated to the description, then all POIs render with "clickable style".

M-Reimer, 2024-01-22:
And some kind of "don't reinvent new output fields" command line switch wouldn't possibly be a feature that could be useful to others?

M-Reimer, 2024-01-22:
Thanks for the information. Then GPSbabel is not the right tool to solve my problem.

closed 2024-01-22 by M-Reimer