GPSBabel/gpsbabel #1264: [feature request] Parsing of Google Takeout TmeLine records.json file

TrueArgus13, 2024-04-07:
Parsing of records.json file will provide detail waypoints from Google takeout timeline. It contains waypoints with additional info.
As GPSBabel already suport other JSON files - it should be very simple to add it (it took about 10 minyutes to wrote script in python). Input may be filtered by date, precision and source (usuallly - GPS/CELL or WIFI)

By the way, is there an option to include precision field in output GPX file?

tsteven4, 2024-04-07:
Support for reading google takeout timelines was added in 1.9.0: https://www.gpsbabel.org/htmldoc-1.9.0/fmt_googletakeout.html.

gpx has an option to change the precision of elevation values, but requires a source code change to alter the precision of other values such as latitude and longitude: https://www.gpsbabel.org/htmldoc-1.9.0/fmt_gpx.html

TrueArgus13, 2024-04-07:
Sorry, probably I was not clear enough. I'm talking not about "Semantic Location History" files which contains timeline objects only (actually - as we can see in TimeLine of Google web), but about raw waypoints, which contained in "Takeout\Location History (Timeline)\Records.json" file. 

Example:
{
  "locations": [{
    "latitudeE7": 324564025,
    "longitudeE7": 350534863,
    "accuracy": 78,
    "altitude": 140,
    "verticalAccuracy": 16,
    "source": "WIFI",
    "deviceTag": 1664314150,
    "platformType": "ANDROID",
    "osLevel": 33,
    "serverTimestamp": "2023-11-24T00:08:40.526Z",
    "deviceTimestamp": "2023-11-24T00:08:40.071Z",
    "batteryCharging": true,
    "formFactor": "PHONE",
    "timestamp": "2023-11-24T00:00:12.137Z"
  }, {
    "latitudeE7": 324564025,
    "longitudeE7": 350534863,
    "accuracy": 100,
    "altitude": 140,
    "verticalAccuracy": 23,
    "source": "WIFI",
    "deviceTag": 1664314150,
    "platformType": "ANDROID",
    "osLevel": 33,
    "serverTimestamp": "2023-11-24T00:08:40.526Z",
    "deviceTimestamp": "2023-11-24T00:08:40.071Z",
    "batteryCharging": true,
    "formFactor": "PHONE",
    "timestamp": "2023-11-24T00:00:32.137Z"
  }, {
....

tsteven4, 2024-04-07:
We would entertain a pull request to support this.

A useful reference is https://locationhistoryformat.com/reference/

postmaxin, 2024-04-07:
I implemented using the "Semantic Location History" folder first because it provides both waypoints and tracks, with trips being broken up intelligently, whereas "Records.json" appears to only provide one gigantic track, your all-time Google location tracking data. Just a track, no place names etc. We could include the Records.json data but there are a few challenges:

* My `Records.json` going from 2012 until last August is 1.2 gigabytes of JSON, and it's one huge "locations" array. We'd want to stream that efficiently so that we don't always have to load the whole thing at once
* We would probably want to correlate coordinates with the tracks in the "Semantic Location History" folder based on time. That way, we can keep waypoints etc and the data that `Records.json` provides is interleaved to produce a more detailed track

TrueArgus13, 2024-04-07:
@postmaxin 
Yes, sometime we need detailed tracks. For example, for geotaggign of photos taken by cameras.
Timestamp filtering is required. "Accuracy" filter is nice-to-have too 😁
For awhile I just parse it to unicsv files (file-per-date) by simple python script and then combine and convert by  GPSBabel.