GPSBabel/gpsbabel #1292: Google Map KLM convertion to GPX - GPX size

Mint-me123, 2024-07-10:
Is it possible to optimize the conversion from a google map KML to a GPX so the GPX stays small. Converting a 500-700 kb KML gives a 2.8 MB GPX. It is for use with a mobile app on a bike so only tracks or routes are needed nothing else. Possible. Perhaps not if google KMLs are just full of "stuff". Downloaded GPX made for bike trips are normally around 200-600 kb so that is why I thought it might be possible.

tsteven4, 2024-07-10:
It may be possible, it depends on what is in your kml that you don't want.  You can use the [nuketypes](https://www.gpsbabel.org/htmldoc-development/filter_nuketypes.html) filter to get rid of waypoints and/or routes and/or tracks.  If you have duplicate track or route data it may be possible to delete it with the [track](https://www.gpsbabel.org/htmldoc-development/filter_track.html) filter.  You can reduce the number of points in a route or track with the [simplify](https://www.gpsbabel.org/htmldoc-development/filter_simplify.html) filter.  I often use that to reduce the number of points recorded by a Garmin Edge to something Google Earth is happier with.

Mint-me123, 2024-07-11:
Hi thank you for your suggestions. I tried the nukes... Filter. Only removing tracks and routes gave me a smaller size but for some reason it showed a different "route". Keeping tracks only or routes only gave a functional "route" but didn't reduce size. I will try to use simplify as well. Perhaps a look inside with a text editor will show me the surplus kb. Thank you again

Mint-me123, 2024-07-12:
Update. Simplify worked a treat. GPS Studio showed that the org kml from Gmaps had 28000 point. Reducing that to 10k, 5k, 2k and finally 1k brought the gpx down to 92 kb and fully functional. Is there a way to see how many points a route has when you elect to simplify like GPS Studio has. Would be a good feature. Thank you for the help. Now I just need to make sure the gpx tracks shows up as a real name and not as track1 2 3 etc... in a text editor i guess.

tsteven4, 2024-07-12:
You can see the number waypoints, routes and tracks by using the validate filter.  I often use it twice in one invocation, once before the simplify filter and once after.

Here is an example.  The fit file contained 1 waypoint and 1 track with 22940 points.  After simplify with an allowed error of 2 meters the track was reduced to 1170 points.

```
$ gpsbabel -i garmin_fit -f 15530281629_ACTIVITY.fit -x validate,debug -x simplify,error=2m -x validate,debug -o gpx -F x.g
px

Processing waypts
point ct: 1, waypt_count: 1

Processing routes
route head ct: 0, route_count: 0
total route point ct: 0, route_waypt_count: 0

Processing tracks
track 0 ct: 22940, waypt_count: 22940
track head ct: 1, track_count: 1
total track point ct: 22940, track_waypt_count: 22940

Processing waypts
point ct: 1, waypt_count: 1

Processing routes
route head ct: 0, route_count: 0
total route point ct: 0, route_waypt_count: 0

Processing tracks
track 0 ct: 1170, waypt_count: 1170
track head ct: 1, track_count: 1
total track point ct: 1170, track_waypt_count: 1170
```

tsteven4, 2024-07-12:
If your intent is to create a route to follow from a track you may wish to also use the transform filter.  The del option deletes the original track after transforming it to a route, the timeless option deletes the track point timestamps.  In this case the transform filter reduced the gpx size an additional 32%.

```
$ gpsbabel -i garmin_fit -f 15530281629_ACTIVITY.fit -x validate,debug -x simplify,error=2m -x transform,rte=trk,del,timeless -x validate,debug -o gpx -F x.gpx && wc x.gpx

Processing waypts
point ct: 1, waypt_count: 1

Processing routes
route head ct: 0, route_count: 0
total route point ct: 0, route_waypt_count: 0

Processing tracks
track 0 ct: 22940, waypt_count: 22940
track head ct: 1, track_count: 1
total track point ct: 22940, track_waypt_count: 22940

Processing waypts
point ct: 1, waypt_count: 1

Processing routes
route 0 ct: 1170, waypt_count: 1170
route head ct: 1, route_count: 1
total route point ct: 1170, route_waypt_count: 1170

Processing tracks
track head ct: 0, track_count: 0
total track point ct: 0, track_waypt_count: 0
  4692   7045 137503 x.gpx
```