GPSBabel/gpsbabel #1305: Trails that were apparently recorded discontinuously on a lowrance unit don't set the trkseg flag.

tsteven4, 2024-07-25:
@bjkowalski can you comment on the possibly discontinuous recording of Trails in lowrance units?

I am debugging a users version 2 lowranceusr file.  It has two "Trails", each with 2000 points.  When I convert it to kml and display in google earth I notice that there are long straight line segments between some of the points.  Sometimes these are hundreds of miles long.   This suggests that the recording of these "Trails" is not continuous.  I see a flag "continuous_flag" in the lowrance code, and the related break option.

We have the concept of a track which is composed of track segments adopted from gpx.   From the gpx 1.1 schema:

>A  Track Segment holds a list of Track Points which are logically  connected in order. To represent a single GPS track where GPS reception  was lost, or the GPS receiver was turned off, start a new Track Segment  for each continuous span of track data.

If I set the new_trkseg flag when the continuous_flag is zero in LowranceusrFormat::lowranceusr_parse_trail then the long straight line segments between some of the points displayed in google earth are gone as one would expect.  There are some other missing segments as well, some of them are between points that are close together and may or may not have been logically connected.

```
       if (!continuous_flag) {<br>
          wpt_tmp-&gt;wpt_flags.new_trkseg = 1;
       }
```

I notice also the lowranceusr writer ignores wpt-&gt;wpt_flags.new_trkseg and has it's own version of continuous.  This seems erroneous as well.

bjkowalski, 2024-07-26:
@tsteven4 I do not recall encountering discontinuous trail recordings but the devices I had access to at the time could not generate USR2 data files.

Looking at the source I am speculating that "continuous" was part of the original implementation which I tried to leave untouched.

Sorry I can't be on any more help.

tsteven4, 2024-07-26:
Thanks for your note @bjkowalski .  It appears the continuous flag is only a part of version 2 and 3 files.  I will submit a PR to educate the lowranceusr version 2 & 3 reader and writer about track segments.

PR #1306 (2024-07-27): teach lowranceusr about track segments
lowranceusr versions 2 & 3 have a continuous flag used to
indicate dicontinuities in a trail.
The lowranceusr version 2 & 3 reader now flags trail discontinuities
as new track segements.
The lowranceusr version 2 & 3 writer now flags new track segments as
discontinuities.

Also fix two subtle bugs:
1) when reading a version 2 or 3 trail with the break option we would
fail to break the trail if the discontinuity was at the start of a
section other than the first section.
2) when writing a version 2 or 3 trail with the merge option if the
number of points e [... cut]
closed 2024-07-27 by tsteven4