GPSBabel/gpsbabel #1383: Google Location Timeline now Device Local - JSON Structure Changed

jbfunk, 2024-11-15:
In the past you could download your Google Timeline data as JSON via Takeout, and the GPSBabel googletakeout format ( https://www.gpsbabel.org/htmldoc-1.9.0/fmt_googletakeout.html ) would parse it correctly.

Recently, Google has transitioned Timeline to a completely local service on a device. You can no longer download Timeline data via Takeout. However, you can still export the data as a JSON file from the device itself in the Android settings menus (under Location>Location services>Timeline>Export Timeline data)--but the structure of the JSON has changed and the GPSBabel googletakeout format does not parse it correctly.

More info on this and some specifics of the new JSON structure can be found here: https://github.com/CarlosBergillos/LocationHistoryFormat/issues/13

Hopefully the googletakeout format can be updated or a new format can be made to support the new device-local Timeline JSON.

postmaxin, 2024-11-15:
Ack!
I was planning on refreshing my timeline download soon too.
I'm super busy with work over the next week but I will take a look if I can find the time.

jbfunk, 2024-11-15:
Regarding the sunset of the Takeout option, I'm not sure if that was a hard date that affected everyone all at once or if it's a rolling-type thing with different users having different deadlines... So it may still be available for some. In any case, at some point in the not too distant future it will be disabled for everyone (if it hasn't been already). But yes, "ack" indeed.

postmaxin, 2024-11-16:
It might make sense to keep the old parser around as "oldgoogletakeout" or
something, so that people can still use data they have already pulled...
then maybe deprecate that in a year or two