GPSBabel/gpsbabel #1391: Same output, exclude=1 or not

jidanni, 2024-12-01:
How could I get the same output for both exclude=1 and not?
```
$ gpsbabel -i gpx -f gpx/tai8to4corners.gpx -o gpx -F 328.gpx -x polygon,file=/tmp/polygon
$ gpsbabel -i gpx -f gpx/tai8to4corners.gpx -o gpx -F tu_main.gpx -x polygon,file=/tmp/polygon,exclude=1
wc 328.gpx tu_main.gpx gpx/tai8to4corners.gpx | sed $\d
  1681   2809  53437 328.gpx
  1681   2809  53437 tu_main.gpx
  1685   2821  42842 gpx/tai8to4corners.gpx
$ cat /tmp/polygon
24.18577 121
24.18577 120
89 120
24.18577 121
```
[tai8to4corners.gpx.gz](https://github.com/user-attachments/files/17969188/tai8to4corners.gpx.gz)
GPSBabel Version 1.9.0.
No there are no actual examples of `exclude` usage on
https://www.gpsbabel.org/htmldoc-1.9.0/filter_polygon.html
there at the bottom. Maybe my triangle wasn't expected.
Maybe it wants squares or more sided objects. I didn't test.

tsteven4, 2024-12-01:
1. the polygon filter works on waypoints, not tracks or routes.
2. command line is processed left to right, filtering after output is useless (https://www.gpsbabel.org/htmldoc-1.9.0/Invocation.html, https://www.gpsbabel.org/htmldoc-1.9.0/Advanced_Usage.html)
```
gpsbabel -i gpx -f ./tai8to4corners.gpx -x transform,wpt=trk,del -x polygon,file=/tmp/polygon,exclude -o gpx -F 328e.gpx
gpsbabel -i gpx -f ./tai8to4corners.gpx -x transform,wpt=trk,del -x polygon,file=/tmp/polygon -o gpx -F 328.gpx
```

closed 2024-12-05 by tsteven4

jidanni, 2024-12-22:
> 1. the polygon filter works on waypoints, not tracks or routes.

Maybe https://www.gpsbabel.org/htmldoc-development/filter_polygon.html could mention that. Else some people might think it also acts on the nodes in tracks  or routes.

> 2. command line is processed left to right, filtering after output is useless (https://www.gpsbabel.org/htmldoc-1.9.0/Invocation.html, https://www.gpsbabel.org/htmldoc-1.9.0/Advanced_Usage.html)

Maybe a warning could be printed if it is detected that the user has done that.