GPSBabel/gpsbabel #1402: Fix documentation: Qt6 not Qt5 is now required

ljb279, 2025-01-23:
Please fix: https://www.gpsbabel.org/htmldoc-1.10.0/Source.html
and https://www.gpsbabel.org/htmldoc-development/Source.html
and xmldoc/chapters/build.xml

These "Building from source" pages say you require Qt-5.12 or newer. This is incorrect. Starting with gpsbabel-1.10.0 you now require Qt-6.2 or newer.

I found this out when I tried and failed to upgrade gpsbabel from 1.9.0 to 1.10.0 on my up-to-date Linux system, which includes Qt-5.15.3.

tsteven4, 2025-01-23:
Oops.  Sorry for the misdirection.  It will take a while for the update documents to appear at gpsbabel.org.  But here is what they will say:

[image]

tsteven4, 2025-01-23:
<p>There are external requirements for bulding.</p><div class="variablelist"><dl class="variablelist"><dt><span class="term">
          <a class="link" href="https://contribute.qt-project.org/" target="_top">Qt</a>
        </span></dt><dd><p>
Qt version 6.2 or newer is required for all builds.
You may download directly from Qt using the <a class="link" href="https://doc.qt.io/qt-6/get-and-install-qt.html" target="_top">Qt supplied directions</a>.
MacOS users can download binaries from <a class="link" href="https://formulae.brew.sh/formula/qt" target="_top">Homebrew</a>.
Fedora or CentOS users may need to 'dnf install qt6-qtbase-devel qt6-qt5compat-devel'.  When in doubt, 'dnf search qt' or 'dnf search qt6' may help you find the correct package name. Ubuntu users may need to 'apt-get install qt6-base-dev qt6-5compat-dev'. Additional qt6 packages are required to build the GUI.  Package names
and versions in Linux frequently change, so you may need to ask your Linux vendor
for help or look in our <a class="link" href="https://github.com/GPSBabel/gpsbabel/tree/master/tools" target="_top">Docker files</a> for inspiration from our automated builds that
use Docker.
  </p></dd><dt><span class="term">
          <a class="link" href="https://libusb.info/" target="_top">libusb 1.0</a>
        </span></dt><dd><p>
is needed to communicate with use with older USB Garmins.
For macOS, we use an included copy.
Fedora users may need to 'dnf install libusb1-devel'.  Ubuntu users may
need to 'apt-get install libusb-1.0-0-dev' or look in our <a class="link" href="https://github.com/GPSBabel/gpsbabel/tree/master/tools" target="_top">Docker files</a> for inspiration from our automated builds that
use Docker.
</p>

PR #1403 (2025-01-24): update building from source directions.
This resolves #1402.  

@robertlipe I can generate the updated documents on github.
closed 2025-01-24 by tsteven4