GPSBabel/gpsbabel #1411: Feature request: support for 70Mai dashcam .txt files

borismasis, 2025-02-18:
Hi there, quick feature request: it would be awesome to add support for 70Mai dashcams which are quite popular now and store GPS data in .txt files. There is a description of how this data is stored and sample code to parse it available at https://forum.mapillary.com/t/using-the-70mai-a810-dashcam-for-mapillary/9130/7?u=boris

An example .txt file from the camera is at https://drive.google.com/file/d/1UUi0qFvFQnkOaBKHbgew3FxDr-4Q2rrJ/view

Thank you for your consideration!

tsteven4, 2025-02-18:
Our [xcsv format](https://www.gpsbabel.org/htmldoc-development/fmt_xcsv.html) gets you part way there.  However, it takes extra work to

- throw out the invalid data (satellite status indicator V, or location 0.000000,0.000000)
- handle the last partial line in your file
- handle the new track indicators
- handle speed (the scaling is supported)
- handle course (the scaling is not supported)

The two greps handle the first point.  The head handles the second point. The xcsv format can handle new track indicators as a field on each line, but it can't handle them when they are separate lines.   awk takes care of the third point, turning the $V02 lines which indicate the start of a new track into a final field in the data lines.  speed and course are ignored.  

```
grep -v ",V," GPSData000001.txt | grep -v ",0.000000,0.000000," | head -n -1 | awk  -v RS="\r\n" '/\$V02/{new=1; next}{printf "%s,%d\n",$0,new; new=0}' | ~/work/rc/bld/gpsbabel -t  -i xcsv,style=mai.style -f - -o kml,points=0 -F GPSData000001.kml
```
You can synthesize speed and course with the track filter.  But if you don't need that just ignores those fields as above.  The required [style file](https://www.gpsbabel.org/htmldoc-development/Styles.html) is included in the zip below.  Also included is the kml file created by the above command showing your tracks in Belarus.

Adding support to handle the rough edges would require:
1) convincing us there is a significant user base that would benefit.  Each format entails significant ongoing maintenance costs so we don't want to add formats that aren't well used.
2) a volunteer to implement the code.
See:
https://www.gpsbabel.org/FAQ.html#new_feature
https://www.gpsbabel.org/FAQ.html#feature_scope

[maidashcam_1411.zip](https://github.com/user-attachments/files/18848435/maidashcam_1411.zip)

borismasis, 2025-02-19:
Thank you! This is very helpful, appreciate such a quick reply! The .kml file you produced looks super to me!

In terms of next steps:

What would be helpful in terms of convincing that a significant user base would benefit? You can see the discussion in the [mapillary forum for users requesting this
](https://forum.mapillary.com/t/using-the-70mai-a810-dashcam-for-mapillary/9130/1) as well as in the [russian language forum](https://4pda.to/forum/index.php?showtopic=1069477) with 285 pages of comments (!). I think part of the reason for the popularity might be that the 70mai cameras are very cheap. On Amazon its only $39 and the [#10 best seller](https://www.amazon.com/s?k=dashcam&s=exact-aware-popularity-rank) for dashcams. 70mai says "[70mai achieved No.1 market share in Russia, Malaysia and Thailand](https://en.70mai.com.cn/about/?utm_source=chatgpt.com)" with 2 million dashcams sold.

Unfortunately I can't personally help with the implementation, but very much appreciate your consideration!

tsteven4, 2025-02-24:
@borismasis if you are still interested in 70mai format support please respond to https://github.com/GPSBabel/gpsbabel/pull/1412#issuecomment-2672279527

PR #1412 (2025-02-27): add 70mai format
This resolves #1411.

This format doesn't quite work standalone with xcsv as discussed in the issue referenced above.  These issues are overcome by creating a custom format.  The format can only read tracks.